fairecasoimeme/ZiGate #104: Trådfri remote

ISO-B, 2018-12-19:
Have anyone tried to debug what it sends to lights? I tried to use zigate with sniffer.bin but ubiqua doesn't seem to work. Also tried to use remote touchlink to zigate without any progress. With fullfunction firmware with router mode remote and zigate changed one message successfully. 

I wonder would it be even possible to use touchlink to zigate? If I am not misunderstood deCONZ managed to get remote commands to server and reprogram them. Could we do same?

pipiche38, 2018-12-19:
I suspect that for the TouchLink we need the Network Formation (which is not implemented as far as I saw)

Right now, what is working very well is using the remote to define groups which then can be piloted directly from the Zigate

ISO-B, 2018-12-19:
What ia easiest way to check what groups device belongs or find all available groups?