fairecasoimeme/ZiGate #105: Building ZiGate firmware

ISO-B, 2018-12-20:
# Installing SDK and using it to build firmware
Figuring how to install SDK and build ZiGate firmware took me some time. I hope this saves someones time and helps developing ZiGate.
## Install BeyondStudio
BeyondStudio is Eclipse IDE for NXP JN51xx C/C++ developers. It is needed to build ZiGate firmware. It is available on NXP website you just need to have account which you can create for free.
1. Download BeyondStudio for NXP(JN-SW-4141) from [HERE](https://www.nxp.com/products/wireless/proprietary-ieee-802.15.4-based/zigbee/zigbee-3.0:ZIGBEE-3-0)
2. Instal BeyondStudio. I recomend to install it to default installation path, but you can install it anywhere. Install it to other place and you will have to replace some paths later.
3. Start BeyondStudio and create workspace when asked. Workspace path should be BeyondStudio-installation-path/workspace
4. Continue to latest released or development version section

## ZiGate latest released version
1. Download JN516x ZigBee 3.0 SDK (JN-SW-4170) for Zigate from [HERE](https://github.com/fairecasoimeme/ZiGate/releases)
2. Extract JN-SW-4170 folder from zip to BeyondStudio-installation-path\sdk\
3. Download ZiGate source code (ZiGate.zip) from [HERE](https://github.com/fairecasoimeme/ZiGate/releases)
4. Extract content of ZiGate.zip to BeyondStudio-installation-path\workspace\ZiGate\
5. Now you should have BeyondStudio-installation-path\workspace\ZiGate\ folder with Build, Doc, JN516x, JN517X, Source, Tools folders and VERSION.txt file
6. Go to section using BeyondStudio

## ZiGate latest development version
1. Download latest development version from Github. Select desired branch and download it. At time of writing this newest branch is v3.0f. https://github.com/fairecasoimeme/ZiGate/archive/v3.0f.zip
2. Extract ZiGate-3.0f\Module Radio\Firmware\src\ZiGate\ folder from zip to BeyondStudio-installation-path\workspace\
3. Now you should have BeyondStudio-installation-path\workspace\ZiGate\ folder with Build, Doc, JN516x, JN517X, Source, Tools folders and VERSION.txt file
4. Extract ZiGate-3.0f\Module Radio\Firmware\src\sdk\JN-SW-4170\ folder from zip to BeyondStudio-installation-path\sdk\

Evrything would be good but Github breaks code line endings. We need to fix those.
1. Executables
	1. Download JN516x ZigBee 3.0 SDK (JN-SW-4170) for Zigate from [HERE](https://github.com/fairecasoimeme/ZiGate/releases)
	2. Extract JN-SW-4170\Tools\ folder from zip to BeyondStudio-installation-path\sdk\JN-SW-4170\ and overwrite all duplicates.
2. Find and remove all \r characters from .mk, .c, .h files
	1. You can use whatever program you want but I will demonstrace it using Notepad++
	2. Open Notepad++
	3. From menu Search -> Find in files
	4. Fill values
		- Find what: \r
		- Leave Replace with empty
		- filters: *.h
		- Path: BeyondStudio-installation-path\sdk\JN-SW-4170\
		- Search mode: Extended
		- Only checkbox that needs to be checked is "In all sub-folders"
	5. Replace in Files. When asked are you sure double check path and if its correct select yes.
	6. Repeat replacing part with all required file endings.
3. Library folders
	1. Replace library folders content with working files from JN-SW-4170.zip
	2. At least following folders need to be replaced
		- /Components/Library/
		- /Platform/DK2/Library/
		- /Platform/DK3/Library/
		- /Platform/DK4/Library/
4. Now everything should be ready for building firmware

## Using BeyondStudio
1. Start BeyondStudio if it is not still running. Select same workspace as earlier
2. From menu select File -> Import
3. From Import popup. Under General select Existing Projects into Workspace. Click next.
4. Select root directory and browse to BeyondStudio-installation-path\workspace\zigate\
5. Now you should see two projects on projects area. Select JN516x one if you are using ZiGate. Click finish.
6. Close welcome window
7. If you installed BeyondStudio anywhere else than default location from Project Explorer select JN-SW-4170. Click it with right mouse button and select properties. Click Edit... button from Location row and find your BeyondStudio installation location from there select sdk and JN-SW-4170. Press Ok. Press Ok.
8. Click your project "JN-AN-1216-Zigbee-3-0-IoT-ControlBridge" and select Build Configurations -> Set active and select desired build. I have used  ZigbeeNodeControlBridge_Coordinator_115200[JN5168] for ZiGate.
9. Click your project again and select Build Configurations -> Build Selected.
10. Make sure that desired configuration is selected and also clean and build are checked. Press ok to start building. This will take some time.
11. If no error other than Memsize.exe appears then you should find your firmware file from folder BeyondStudio-installation-path\workspace\zigate\build\ZigbeeNodeControlBridge\ it is named acording your build configuration and ends with .bin

pipiche38, 2018-12-20:
Very useful contribution!

pipiche38, 2019-01-07:
@ISO-B  Thanks very much for this step by step help.

May be you could help me, while
A file : `C:\NXP\bstudio_nxp\workspace\ZiGate\Build\ZigbeeNodeControlBridge\ZigbeeNodeControlBridge_JN5169_COORDINATOR_115200.bin
/JN-AN-1216-Zigbee-3-0-IoT-ControlBridge/Build/ZigbeeNodeControlBridge/ZigbeeNodeControlBridge_JN5169_FULL_FUN
` as been generated, I'm however having such error message.

/c/NXP/bstudio_nxp/sdk/Tools/ba-elf-ba2-r36379/bin/ba-elf-objcopy -j .version -j .bir -j .flashheader -j .vsr_table -j .vsr_handlers -j .ro_mac_address -j .ro_ota_header -j .ro_se_customData -j .rodata -j .text -j .data -j .bss -j .heap -j .stack -S -O binary ZigbeeNodeControlBridge_JN5168_FULL_FUNC_DEVICE_115200.elf ZigbeeNodeControlBridge_JN5168_FULL_FUNC_DEVICE_115200.bin
/c/NXP/bstudio_nxp/sdk/JN-SW-4170/Tools/Memsize/memsize.exe /c/NXP/bstudio_nxp/workspace/ZiGate/Build/ZigbeeNodeControlBridge/ZigbeeNodeControlBridge_JN5168_FULL_FUNC_DEVICE_115200.elf /c/NXP/bstudio_nxp/workspace/ZiGate/Build/ZigbeeNodeControlBridge/ZigbeeNodeControlBridge_JN5168_FULL_FUNC_DEVICE_115200_NOLTO.map ZigbeeNodeControlBridge.json
Error loading Python DLL 'C:/Users/pichon/AppData/Local/Temp\_MEI6282\python36.dll'.
LoadLibrary: The parameter is incorrect.
```

ISO-B, 2019-01-07:
Binary is correct.

That error is expected since NXP's JN-SW-4170 has broken memsize.exe . Only thing that memsize.exe is used is to generate xls file that gives overview how firmware uses memory. You can remove memsize.exe command from Makefile then you don't get that error.

KiwiHC16, 2019-01-23:
Don't mix envirronement (Linux, OSX, Win) or you will waste hours and hours.
Follow ISO-B guide from A to Z being in Win7.

KiwiHC16, 2019-01-24:
As the compilation is closely linked to the folders organisation here is a solution I use to ease the dev. 

First create a folder link with the command:
mklink /D C:\NXP\bstudio_nxp\workspace\ZiGate C:\NXP\bstudio_nxp\ZiGate\ModuleRadio\Firmware\src\ZiGate

You need to be admin to do so:
[image]

Then we will use GitHub to get the latest version. Start GitHub Desktop and "Clone a repository" from URL: https://github.com/fairecasoimeme/ZiGate.git
[image]

KiwiHC16, 2019-01-24:
Let's start NXP IDE Studio, use default workspace or define

 it up:
[image]

KiwiHC16, 2019-01-24:
Time to compile:
[image]

and you should get this error:
[image]

So let's clean first:
[image]

Clic on little green target:
[image]

KiwiHC16, 2019-01-24:
As a result you should have a target with clean key word:
[image]

Clic on clean to clean...
[image]

Time to compile:
[image]
Clic on little hammer.

KiwiHC16, 2019-01-24:
[image]

KiwiHC16, 2019-01-24:
We compiled the master branche, let's compile the branche 3.0.f.
Back to github (close NXP Studio by default when you use GitHub to avoid conflicts)

First discarde the changes and come back to master:
[image]

Select 3.0.f branche:
[image]

Then start

ISO-B, 2019-01-24:
One thing that would also help, if it's ok for @fairecasoimeme, to fix all those line-endings and commit files to git

fairecasoimeme, 2019-01-24:
You're right,
I'll try to solve this issue this week-end ...

fairecasoimeme, 2019-01-27:
done. Please tell me of it's ok for you. I'm newbie in github

fairecasoimeme, 2019-01-30:
Ok, I'll do that tomorrow
Thx !

handfreezer, 2019-02-10:
Just to tell this issue is very usefull! Thanks!

pipiche38, 2019-03-06:
It might be good that issue is translated into a web page for documentation purposes

badzz, 2020-01-31:
> Hi everyone
> I get a problem When I build ZiGate v3.1a (final) project.
> Use Zigate from https://pypi.org/project/zigate/ to connect to Coordinator,
> 
> * Use ZiGate_coordinator_JN5169.bin (in forder bin) it connect ok.
> * Build JN5169-115200
>   [image]
>   => I can't open port to connect Coordinator
>   Can you help me? I don't know what reason !!!

you have to use 5168 not 5159

badzz, 2020-04-25:
If you want to directly work in beyondstudio on your github repository, do the following on a already installed config

stop beyondstudio for NXP
delete, rename or move somewhere else the folders c:\nxp\bstudio_nxp\sdk\JN-SW-4170 and c:\nxp\bstudio_nxp\workspace\ZiGate

if your git repo is under c:\workfolder\ZiGate, then do the following

mklink /J c:\nxp\bstudio_nxp\sdk\JN-SW-4170 "C:\workfolder\ZiGate\Module Radio\Firmware\src\sdk\JN-SW-4170"
mklink /J c:\nxp\bstudio_nxp\workspace\ZiGate "C:\workfolder\ZiGate\Module Radio\Firmware\src\ZiGate"

after checking out branches, even more for older ones you will to make sure that your obj files are cleaned. In BeyondStudio for NXP , you will have to do Project -> Clean 

@fairecasoimeme : should we move all this to a wiki page ?

closed 2020-09-30 by fairecasoimeme

pipiche38, 2020-09-30:
https://github.com/fairecasoimeme/ZiGate/wiki/GUIDE:-Building-ZiGate-firmware