fairecasoimeme/ZiGate #107: Compatibility with LivingColors Gen3 remote ?

Cactusbone, 2018-12-23:
[image]

I could not get the LivingColors Gen 3 remote (picture) to pair with zigate on jeedom

Reset procedure should be press 1 button and single dot button at once 
[image]

I'll try to get the ZGUI feedback

Anyone managed to pair it successfully ?

Cactusbone, 2018-12-23:
Ok, I managed to get it to pair, by entering in inclusion mode, and pressing the inner reset button for 5sec (until the led light turned on)

[image]

Cactusbone, 2018-12-23:
Could not get TestGUI to work :(

I believe it should be possible to link the remote to zigate without reset by mimicking the remote link feature 

[image]

(From manual at https://www.download.p4c.philips.com/files/7/7099860ph/7099860ph_dfu_rus.pdf)

Smanar, 2018-12-23:
Just for information the tradfri remote can't be paired to zigate yet https://github.com/fairecasoimeme/ZiGate/issues/64
If this one works same, I think it's normal having same result.

Cactusbone, 2018-12-23:
I managed to pair the tradfri dimmer remote though, which remote are you talking about ?

pipiche38, 2020-09-30:
This issue has been closed due to no activity since while!

closed 2020-09-30 by fairecasoimeme