fairecasoimeme/ZiGate #109: USB write/read error

nherreyre, 2018-12-26:
Hello,

je rencontre depuis 3 jours le meme probleme  que #41 (alors que tout marchait plutot bien) a savoir:

Dec 26 22:10:12 localhost kernel: [812606.516971] pl2303 ttyUSB0: pl2303_set_control_lines - failed: -110
Dec 26 22:10:12 localhost kernel: [812606.620966] pl2303 ttyUSB0: error sending break = -110
Dec 26 22:10:12 localhost kernel: [812606.724971] pl2303 1-1.4:1.0: pl2303_vendor_write - failed to write [0008]: -110
Dec 26 22:10:12 localhost kernel: [812606.828965] pl2303 1-1.4:1.0: pl2303_vendor_write - failed to write [0009]: -110
Dec 26 22:10:12 localhost kernel: [812606.932968] pl2303 ttyUSB0: pl2303_get_line_request - failed: -110
Dec 26 22:10:12 localhost kernel: [812607.036959] pl2303 ttyUSB0: pl2303_set_line_request - failed: -110
Dec 26 22:10:13 localhost kernel: [812607.140954] pl2303 1-1.4:1.0: pl2303_vendor_write - failed to write [0000]: -110
Dec 26 22:10:13 localhost kernel: [812607.244958] pl2303 ttyUSB0: pl2303_set_control_lines - failed: -110
Dec 26 22:10:13 localhost kernel: [812607.349010] pl2303 ttyUSB0: pl2303_set_control_lines - failed: -110

toutes les minutes.

j'ai essayé de:
- rebooter l'OdroidC2, 
- deplug/plug le module USB (plus de 10 fois, avec et sans le module zigate)
- de brancher le clé sur un windows -> il me met une erreur que le driver ne peut démarrer avec un point d'interogation
(PI je n'uilise pas de rallonge USB)

Pensez vous que le module prolific est HS ?

Merci,
Cdlt,
Nicolas Herreyre.
Commande n°1823 (30 mai 2018)

fairecasoimeme, 2018-12-27:
Bonjour,

Il est possible que la clef USB TTL soit endommagé. Je vous en envoie 2 autres (au cas où). vous recevrez un mail de confirmation.

Fred.

closed 2018-12-27 by fairecasoimeme

nherreyre, 2019-01-07:
Bonjour,

effectivement c'etait ca, le nouveau module TTL resout le probleme.

Merci!