fairecasoimeme/ZiGate #110: Télécommande TRUST 71171 reconnu en générique

shupa, 2018-12-27:
Hello,
Tout d'abord super boulot à toi !

J'arrive bien à inclure la télécommande mais elle n'est pas reconnu comme tel (plutôt comme un objet générique) et je ne possède donc aucune commande exploitable.
[image]
[image]

Elle est pourtant indiqué comme compatible (https://zigate.fr/produits-trust-compatibles-zigate/) donc je suppose que j'ai loupé quelque chose ?
Egalement j'ai régulièrement une notif dans jeedom me demandant de rafraîchir la télécommande.

Une idée ? Y a t'il la possibilité de forcer le type de l'objet ?

D'avance merci !

pipiche38, 2018-12-27:
Je ne reconnais pas l’interface web , mais il est possible que la télécommande se comporte comme celle d’IKEA. C’est-a-did qu’elle peut être appairée, et qu’elle peut piloter les ampoules, gérer un groupe.

shupa, 2018-12-27:
L'interface web est juste jeedom avec le theme "dark sobre".
La télécommande étant compatible zigate je suppose qu'il est donc possible de récupérer l'info lorsqu'on presse un des boutons (on/off, moins de luminosité, plus de luminosité, id de la scene ,...). Non ?
Qu'avez vous comme commandes disponibles dans jeedom/zigate avec une télécommande ikea ?

pipiche38, 2019-01-08:
Je ne sais pas comment tu fais, mais la séquence devrait

1- récupérer le group géré par la télécommande en faisant par exemple un Get Group (membership) sur la lampe qui est associée/pilotée par la télécommande
2- ajouter le groupe membership sur la prise ikea via le Add group

A ce stade tu devrais avoir ajouter cette prise à ton groupe et celle-ci devrait être pilotable par le ZGUI pour un On/off (via le On/off cluster en choisisant le Address Mode group) et aussi par le télécommande ikea

shupa, 2019-01-08:
Oui oui je suis bien cette séquence et ai parfaitement compris le principe.

J'arrive parfaitement à voir/ajouter/supprimer un group sur un philips hue mais sur mes prises IKEA rien n'a faire. Je récupère bien l'ID avec le get devices list et fait le lien avec les détails dans le plugin zigate pour identifier la prise mais les options get group / add / remove n'ont aucun effet dessus.
J'ai également essayé avec une ampoule ikea E14 et même problème sous TestGUI.
Alors que j'arrive bien à commander cette ampoule avec la télécommande donc elle gère les groupes.
J'ai vraiment l'impression que c'est générique au composant IKEA et TestGUI.

Arrivez vous à manipuler les groupes de vos ampoules/prises IKEA avec TestGUI ?

PS: Je suis sous le firmware 3.0d

lboue, 2019-01-08:
Bonjour,
> La seule "vraie" télécommande où on pourra récupérer les ON/OFF est la télécommande philips HUE mais ce sera disponible dans la version 3.0f.

 @fairecasoimeme Est ce que cela signifie que cela fonctionne avec la version ZiGate_3.0f_pre_release.bin du commit 9ea73ad14662cee4ba3ae6c6c8f065798af1ed32 et un Philips Hue Dimmer switch (ref 046677458140).
J'en ai un chez moi pour faire un test ?

Cordialement,

pipiche38, 2019-01-08:
Pour moi ici, ca fonctionne très bien. Pas de prise Ikea, mais des variétés d'Ampoules Ikea Tradfri ( White Color, Color et White) et en y ajoutant une prise Xiaomi.

Aucun problème!

L'ensemble étant piloté au travers d'un groupe.

Cordialement,

fairecasoimeme, 2019-01-08:
@lboue oui mais il y a une procédure à suivre.
1- inclure la télécommande dans le réseau de la ZiGate
2- faire un bind entre la ZiGate et la télécommande hue
[image]

178801021AE3DF - l'adresse de la télécommande Hue
158D0001B22E66 - l'adresse de la ZiGate

3- après appuies sur la télécommande la ZiGate reçoit les informations

Je donnerai plus de détail à la sortie dur firmware. (je rédigerai un petit article à ce sujet)

@shupa : l'erreur provient de ton endpoint, les philips utilisent le endpoint 0x0b alors que les Ikea c'est 0x01

shupa, 2019-01-09:
@fairecasoimeme: Je me disais bien que quelque chose ne tournait pas rond avec mes produits Ikea :)
Un grand merci à toi ça fonctionne évidemment bien mieux avec le bon endpoint...
Je clotûre l'issue qui a déjà pas mal dérivé (en partie de ma faute :) ) !

Merci et encore une fois super boulot

closed 2019-01-09 by shupa