fairecasoimeme/ZiGate #112: Legrand store control not supported ?

dcolens, 2019-01-02:
I'm trying to pair a https://www.legrand.fr/pro/catalogue/42566-version-celiane-with-netatmo/interrupteur-filaire-connecte-pour-volet-roulant-celiane-with-netatmo-blanc with my zigate. The device was not paired to anything else before (I don't own the Legrand gateway), I read the test article you wrote, but this particular device is not covered in it. When I try the pairing using https://github.com/doudz/zigate I get no output, the legrand-store-switch led remains red.

I'm opening this issue following a discussion with @doudz in https://github.com/doudz/zigate/issues/63. I can help with some of the tshooting, unfortunately, I don't have a windows machine around so can't use the win apps.  

- zigate firmware: 3.0e
- host: rpi3
- python version: 3.5

pipiche38, 2019-01-02:
From my knowledge on the Legrand stuff and especially with the Switches, if you take a raw device from the market, then there is no much functionality available. 

Like the switch you get only the On/off and not the Power reporting. You need first to do a Firmware update in order to get features enabled

Smanar, 2019-01-02:
Have you tried with pressing the reset button during 10s ?

dcolens, 2019-01-17:
Thanks for the suggestion, I put the zigate in permit join and then pressed the reset button for 10s, I managed to pair it twice (out of maybe 8 tries). It is not consistent but it is possible to pair it this way!

pipiche38, 2019-03-06:
Is that not cover by #125

dcolens, 2019-03-06:
yes!

closed 2019-03-06 by dcolens