fairecasoimeme/ZiGate #113: Question : Configure reporting request for multiple attributes

doudz, 2019-01-07:
I have a question about reporting request (command 0x0120) for multiple attributes in the same request
The current message structure is :
```
<address mode: uint8_t>
<target short address: uint16_t>
<source endpoint: uint8_t>
<destination endpoint: uint8_t>
<Cluster id: uint16_t>
<direction: uint8_t>
<manufacturer specific: uint8_t>
<manufacturer id: uint16_t>
<number of attributes: uint8_t>
<attributes list: data list of uint16_t each>
Attribute direction : uint8_t
Attribute type : uint8_t
Attribute id : uint16_t
Min interval : uint16_t
Max interval : uint16_t
Timeout : uint16_t
Change : uint8_t
```
But, does it mean that I have to replicate the end of the message for each attribute ?
```
Attribute direction : uint8_t
Attribute type : uint8_t
Attribute id : uint16_t
Min interval : uint16_t
Max interval : uint16_t
Timeout : uint16_t
Change : uint8_t
```
So for example, for 2 attributes, the structure will be :
```
<address mode: uint8_t>
<target short address: uint16_t>
<source endpoint: uint8_t>
<destination endpoint: uint8_t>
<Cluster id: uint16_t>
<direction: uint8_t>
<manufacturer specific: uint8_t>
<manufacturer id: uint16_t>
<number of attributes: uint8_t>
Attribute direction : uint8_t
Attribute type : uint8_t
Attribute id : uint16_t
Min interval : uint16_t
Max interval : uint16_t
Timeout : uint16_t
Change : uint8_t
Attribute direction : uint8_t
Attribute type : uint8_t
Attribute id : uint16_t
Min interval : uint16_t
Max interval : uint16_t
Timeout : uint16_t
Change : uint8_t
```

pipiche38, 2019-01-07:
@doudz that is a good point. While configureReporting attribute by atribute works, I was never able to make it working with multiple attribute

Last, your assumption is correct you should duplicate the attribute structure (at least this is how I understood it)

doudz, 2019-01-07:
Thanks for your comment @pipiche38 but that's why I'm asking, so saying I'm right but it doesn't work, doesn't help...
So please answer only if you really have informations

pipiche38, 2019-01-07:
@doudz quiet surprise on your way to answer !

Just read the documentation https://www.nxp.com/docs/en/user-guide/JN-UG-3077.pdf
page 766 and it will confirm what I said !

doudz, 2019-01-07:
@pipiche38 I'm sorry if offend you.
I've already read the NXP doc but I was just looking for a confirmation and you said that you never be able to make it working.
I'll make some tests

Thank you

doudz, 2019-01-07:
After testing, I confirm that trying to configure more that one attribute doesn't work !
Trying to configure reporting on 2 attributes on my OSRAM bulb make things crazy !
I got periodic device announce ??!!!! (firmware pre 3.0f)

```
2019-01-07 14:54:42 DEBUG (Thread-14) [zigate] Msg to send b'01200022f002a15401030300000000000100200000000100050000000020000100010000000000'
2019-01-07 14:54:42 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-01-07 14:54:42 DEBUG (Thread-14) [zigate] Encoded Msg to send b'01021120021022f00212a15402110213021302100210021002100210021102102002100210021002110210021502100210021002102002100211021002110210021002100210021003'
2019-01-07 14:54:42 DEBUG (ZiGate-Decode data) [zigate] Device Announce
2019-01-07 14:54:42 DEBUG (Thread-14) [zigate] Waiting for status message for command 0x0120
2019-01-07 14:54:42 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_DEVICE_UPDATED
2019-01-07 14:54:42 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15%\x02\x10\x81\x02\x11 \x02\x10\x03'
2019-01-07 14:54:42 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-01-07 14:54:42 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8000: b'00810120'
2019-01-07 14:54:42 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:129, packet_type:288, error:b'', rssi:0
2019-01-07 14:54:42 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-01-07 14:54:42 DEBUG (Thread-14) [zigate] STATUS code to command 0x0120:0
2019-01-07 14:54:42 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x81 \x02\x10\x02\x18\x98\x81\xa1T\x02\x13\x02\x13\x02\x10\x02\x10E\x03'
2019-01-07 14:54:42 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-01-07 14:54:42 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8120: b'81a15403030000'
2019-01-07 14:54:42 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8120 - Configure Reporting response : sequence:129, addr:a154, endpoint:3, cluster:768, status:0, rssi:69
2019-01-07 14:54:42 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-01-07 14:54:43 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x02\x10M\x02\x10\x02\x1c\xd9\xa1T|\xb0>\xaa\x02\x10\xa8\x95\xce\x8eH\x03\x01\x87\x02\x11\x02\x10\x02\x13\x85\x02\x10\x02\x10\x02\x10\x03'
2019-01-07 14:54:43 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-01-07 14:54:43 DEBUG (ZiGate-Decode data) [zigate] Received response 0x004d: b'a1547cb03eaa00a895ce8e'
2019-01-07 14:54:43 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-01-07 14:54:43 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x004D - Device announce : addr:a154, ieee:7cb03eaa00a895ce, mac_capability:10001110, rssi:72
2019-01-07 14:54:43 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8701: b'0000'
2019-01-07 14:54:43 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-01-07 14:54:43 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8701 - Route Discovery Confirmation : status:0, network_status:0, rssi:0
2019-01-07 14:54:43 DEBUG (ZiGate-Decode data) [zigate] Device Announce
2019-01-07 14:54:43 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-01-07 14:54:43 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_DEVICE_UPDATED
2019-01-07 14:54:44 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x81\x02\x12\x02\x10\x02\x1e\x11\x02\x10\xa1T\x02\x13\x02\x13\x02\x10\x02\x10\x02\x10\x02\x10 \x02\x10\x02\x11\x02\x10H\x03'
2019-01-07 14:54:44 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-01-07 14:54:44 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8102: b'00a15403030000000020000100'
2019-01-07 14:54:44 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8102 - Individual Attribute Report : sequence:0, addr:a154, endpoint:3, cluster:768, attribute:0, status:0, data_type:32, size:1, data:0, rssi:72
2019-01-07 14:54:44 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-01-07 14:54:44 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_ATTRIBUTE_UPDATED
2019-01-07 14:54:44 DEBUG (ZiGate-Decode data) [custom_components.zigate] Update attribute for device OSRAM CLA60 RGBW OSRAM (a154) 7cb03eaa00a895ce {'cluster': 768, 'addr': 'a154', 'value': 0, 'name': 'current_hue', 'data': 0, 'attribute': 0, 'endpoint': 3}
2019-01-07 14:54:45 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x81\x02\x12\x02\x10\x02\x1e\x10\x02\x11\xa1T\x02\x13\x02\x13\x02\x10\x02\x10\x02\x10\x02\x10 \x02\x10\x02\x11\x02\x10H\x03'
2019-01-07 14:54:45 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-01-07 14:54:45 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8102: b'01a15403030000000020000100'
2019-01-07 14:54:45 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8102 - Individual Attribute Report : sequence:1, addr:a154, endpoint:3, cluster:768, attribute:0, status:0, data_type:32, size:1, data:0, rssi:72
2019-01-07 14:54:45 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-01-07 14:54:45 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_ATTRIBUTE_UPDATED
2019-01-07 14:54:45 DEBUG (ZiGate-Decode data) [custom_components.zigate] Update attribute for device OSRAM CLA60 RGBW OSRAM (a154) 7cb03eaa00a895ce {'cluster': 768, 'addr': 'a154', 'value': 0, 'name': 'current_hue', 'data': 0, 'attribute': 0, ' [... cut ...]
2019-01-07 14:54:47 DEBUG (ZiGate-Decode data) [zigate] Received response 0x004d: b'a1547cb03eaa00a895ce8e'
2019-01-07 14:54:47 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x004D - Device announce : addr:a154, ieee:7cb03eaa00a895ce, mac_capability:10001110, rssi:72
2019-01-07 14:54:47 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-01-07 14:54:47 DEBUG (ZiGate-Decode data) [zigate] Device Announce
2019-01-07 14:54:47 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_DEVICE_UPDATED

```

doudz, 2019-01-07:
Now I can't control my bulb anymore, I get APS Data confirm Fail with status 212.
Maybe I have to reset the bulb

pipiche38, 2019-01-07:
If I look your frame '01200022f002a15401030300000000000100200000000100050000000020000100010000000000' it looks like the address mode is 0x00. Shouldn't be 0x02 ?

doudz, 2019-01-07:
No. I split the msg :
```
0120 0022f0 02 a154 01 03 0300 00 00 0000 01 00 20 0000 0001 0005 0000 00 00 20 0001 0001 0000 0000 00
```

pipiche38, 2019-01-07:
You right. I mixed the decoding with the way we do in the domoticz plugin. Sorry for that.

Just the Number of attributes is equal to 01 and not 02. 

Also from the NXP documentation: I remember that TimeOut must be higher than the Max reporting Interval

addr mode: 02 
sAddr: a154 
sEP: 01
dEP:  03
Cluster:  0300
Direction: 00
Manuf spe:  00
Manuf Id: 0000
Nb attribute:  01     <=== Shouldn't be 02 ?
    Attr direct: 00
    Attr type:  20
    Attribute:  0000
    MinInter:  0001
   Max Inter: 0005
   TimeOut: 0000
   Change: 00 
   
   Direction: 00 
   Type: 20 
   Attribute: 0001 
   MinInterval: 0001 
   MaxInterval: 0000 
   TimeOut: 0000 
   Change: 00

doudz, 2019-01-07:
yes you're right, nb of attribute should be 2, since I made some manual tests, I probably don't put the right number.

doudz, 2019-01-07:
About the timeout, a value higher than max reporting interval is recommended, except that a value of 0 means disable timeout

pipiche38, 2019-01-07:
Now, I rethink of your issue. As you had only 01 as attribute number it should have behave as sending one attribute. And shouldn't have generated the issue you had ...

pipiche38, 2019-01-07:
Did some test on my side, and here is the paylod associated to the 0x0120 command

028c9f010103000000117c06002100070384012C0FFF010020000003840E100FFF010020000103840E100FFF010021000303840E100FFF010021000403840E100FFF010030000803840E100FFF01

I'm getting a status 0x15 in return of the command !

In conclusion only 0x0120 command with 1 attribute are succeeding !

doudz, 2019-01-09:
0x15 = Invalid destination EndPoint ?
Additionnally you set a manufacturer code 117c but you set manufacturer specific to 00 which means no manufacturer code

pipiche38, 2019-01-09:
@doudz thanks for watching it. Did you succed on your side to send Multiple Attribute ?

 I'll dig in that direction, but same Ep same Manufacturer Code and Specif works attribute by attribute

Here the Device is an TRADFRI bulb E27 WS clear 950lm ,whcih has only 1 EndPoint and as regards to the frame is correctly set ...

pipiche38, 2019-01-11:
@doudz , thanks very much ... you pointed indeed  the right issue on the Manufacturer Id .
Funny enough when sending attribute by attribute it seems that working ...

Anycase, multiple attributes configureReporting works .

Big thanks

ISO-B, 2019-03-22:
Is this still problem in v3.0f?

pipiche38, 2019-07-26:
@ISO-B  I don't think so. I do not have any issue on my end. But I think @doudz one was different !

doudz, 2019-07-26:
I've to check

pipiche38, 2020-01-27:
This is not a BUG, the firmware behave correctly. I however consider that the API documentation is not accurate , as the Change field is not always an uint8.

The Change field size depends on the dataType

pipiche38, 2020-06-29:
The issue is addressed by #308

fairecasoimeme, 2020-09-30:
Fixed on version v3.1d

closed 2020-09-30 by fairecasoimeme