fairecasoimeme/ZiGate #115: Does write attribute request support multiple attributes ? (command 0x0110)

doudz, 2019-01-07:
Documentation says that write attribute request supports multiple attributes writing in the same request (attributes list)
But looking at firmware code I don't think so
https://github.com/fairecasoimeme/ZiGate/blob/master/Module%20Radio/Firmware/src/ZiGate/Source/ZigbeeNodeControlBridge/app_Znc_cmds.c#L1840

Can you confirm ?

fairecasoimeme, 2019-01-09:
I confirm :(

doudz, 2019-01-30:
So, could you just update documentation ?

ISO-B, 2019-03-08:
I don't think that documentation need update.  Case that handles E_SL_MSG_WRITE ATTRIBUTE_REQUEST send data to APP_eSendWriteAttributesRequest. 
https://github.com/fairecasoimeme/ZiGate/blob/76c328bb44a6fdd1774c86807b101c9c15edfebc/Module%20Radio/Firmware/src/ZiGate/Source/ZigbeeNodeControlBridge/app_Znc_cmds.c#L3504-L3624

For me it seems that it should be able to handle attributes list, see lines 3585-3603.

fairecasoimeme, 2020-09-30:
won't fix because not necessary

closed 2020-09-30 by fairecasoimeme