fairecasoimeme/ZiGate #118: Response 0x8120 should provide attribute_id

doudz, 2019-01-09:
Currently the response provide, short address, endpoint id , cluster id but not attribute_id
Having it could help a lot to find attribute that don't support reporting

fairecasoimeme, 2019-01-09:
Ok, I will add attribute id but I don't know in which condition I can test the function.
Please tell me with which device I can test

doudz, 2019-01-09:
a good candidate is a IKEA Bulb, endpoint 1, cluster 6, attribute 0

fairecasoimeme, 2019-01-16:
Ok It's seems to be ok for me. Added to v3.0f

pipiche38, 2019-01-31:
In the new version, what exactly is the behaviour as I see that it is updated in zigate.fr
Do we have the list of Attribute which are only when the Status is != 0, but then what do we have when Status is 0 ?

I'm having some difficulties to understand and to see how to decode it , especially as the Attribute are before the status

<Sequence number: uint8_t> 
<Src address : uint16_t> 
<Endpoint: uint8_t> 
<Cluster id: uint16_t> 
<Attribute Enum: uint16_t> (add in v3.0f) 
 <Status: uint8_t>

ISO-B, 2019-03-08:
When you have multiple attributes you will get single E_SL_MSG_CONFIG_REPORTING_RESPONSE response for each attribute. Response has attribute_id and status for that attribute. Order of the parameters should not make it hard to decode it since attribute_id is always u16 and status is u8.

ISO-B, 2019-03-22:
Feature is implemented and documentation is updated.

closed 2019-03-22 by ISO-B