fairecasoimeme/ZiGate #121: Zigate sur hub usb alimenté + clé Z-wave Aeotec gen5 = zigate failed to connect

Amaurgit, 2019-01-14:
Salut à tous,

Je rencontre un problème avec mon installation jeedom :

- Tout est ok quand j'ai seulement ma zigate sur le hub
Lorsque je branche ma Aeotec gen5 sur le même hub usb aukey (alimentation externe), la zigate se déconnecte. Est-ce que je fais quelque chose de travers ? J'ai l'impression que le port saute lorsqu'un autre périphérique arrive sur la même "ligne usb".

Merci du coup de pouce :)

pipiche38, 2019-01-16:
c'est pas un problème de configuration udev ?

Amaurgit, 2019-01-16:
Ah, tu veux dire que mon mapping de ports USB change au fil de l'eau ?
Comment je peux les persister sur le hub ?

Je ne suis clairement pas un expert des périphériques sur linux :)
Merci !

doudz, 2019-01-16:
Que donne les commandes lsusb et  ls /dev/serial/by-id

Amaurgit, 2019-01-16:
Juste la ZiGate sur le hub :
```
lsusb =>
Bus 001 Device 004: ID 152d:1561 JMicron Technology Corp. / JMicron USA Technology Corp.
Bus 001 Device 008: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port
Bus 001 Device 007: ID 2109:2813 VIA Labs, Inc.
Bus 001 Device 006: ID 2109:2813 VIA Labs, Inc.
Bus 001 Device 005: ID 0424:7800 Standard Microsystems Corp.
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
ls /dev/serial/by-id =>
usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0
```
ZiGate + Aeotec Gen5 : 

```
lsusb =>
Bus 001 Device 004: ID 152d:1561 JMicron Technology Corp. / JMicron USA Technology Corp.
Bus 001 Device 011: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port
Bus 001 Device 010: ID 2109:2813 VIA Labs, Inc.
Bus 001 Device 009: ID 2109:2813 VIA Labs, Inc.
Bus 001 Device 005: ID 0424:7800 Standard Microsystems Corp.
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
ls /dev/serial/by-id =>
usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0
```
Juste Aeotec Gen5 : 

```
lsusb =>
Bus 001 Device 004: ID 152d:1561 JMicron Technology Corp. / JMicron USA Technology Corp.
Bus 001 Device 010: ID 2109:2813 VIA Labs, Inc.
Bus 001 Device 009: ID 2109:2813 VIA Labs, Inc.
Bus 001 Device 005: ID 0424:7800 Standard Microsystems Corp.
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
ls /dev/serial/by-id =>
ls: cannot access '/dev/serial/by-id': No such file or directory
```

Je suis un peu perplexe :)

z3tof, 2019-01-16:
Tu peux créer automatiquement un alias sur le bon tty suivant son id de vendor et de product.

Pour moi qui ai un module zigate et un module rfxcom, jai créé un fichier "/etc/udev/rules.d/10-USB-Domoticz.rules" avec :

SUBSYSTEM=="tty", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", SYMLINK+="ttyUSB-RFXCom"
SUBSYSTEM=="tty", ATTRS{idVendor}=="067b", ATTRS{idProduct}=="2303", SYMLINK+="ttyUSB-ZIGATE"

Comme ca, toujours le même nom de tty...

Amaurgit, 2019-01-17:
1) J'ai réinstallé mon pi hier et toujours le même problème
2) J'ai parcouru pas mal de topic et je rencontrais quelques problèmes de boot sur un SSD 2,5". Je vais tenter de repasser sur un mSata et de refaire une install propre. 
3) Je vous tiens au courant :)

@pdecat tes dongles sont sur hub USB avec alimentation externe également ?

pdecat, 2019-01-17:
> @pdecat tes dongles sont sur hub USB avec alimentation externe également ?

Les deux sur le même hub, mais pas d'alimentation externe.

Note : je ne déploie pas hassio sur Rasberry Pi mais dans une [machine virtuelle avec KVM/Qemu](https://www.home-assistant.io/blog/2017/11/29/hassio-virtual-machine/) et partage des deux devices USB de l'hôte.

Amaurgit, 2019-01-21:
De retour !
Fin mot de l'histoire : problème de SSD ! Je viens de passer sur un msata et tout fonctionne, les deux démons tournents de concert et sur le même hub USB. Merci à tous :)

closed 2019-01-21 by Amaurgit