fairecasoimeme/ZiGate #122: Bind response - 0x80

pipiche38, 2019-01-16:


pipiche38, 2019-01-16:
Would it be possible to have the source address in the bind response ?
either the IEEE or Short address would be welcome

ISO-B, 2019-01-17:
Adding following lines
```
ZNC_BUF_U8_UPD ( &au8LinkTxBuffer [u16Length], psStackEvent->uEvent.sApsDataIndEvent.u8SrcEndpoint, u16Length );
ZNC_BUF_U8_UPD ( &au8LinkTxBuffer [u16Length], psStackEvent->uEvent.sApsDataIndEvent.u8SrcAddrMode, u16Length );
```
after
https://github.com/fairecasoimeme/ZiGate/blob/d18f5e7e82f0660f909d144d217198f0468be072/Module%20Radio/Firmware/src/ZiGate/Source/ZigbeeNodeControlBridge/app_general_events_handler.c#L852

should do the job. I can't see any reason why this couldn't be added.

pipiche38, 2019-01-17:
@ISO-B Are you in keybase.io ? Would like to chat in MP

pipiche38, 2019-02-13:
@ISO-B , @fairecasoimeme  
it seems that it has not been implemented in 3.0f , correct ?

pipiche38, 2019-03-06:
@ISO-B if you can make this one in 3.10, this would be great!

ISO-B, 2019-03-07:
This will be in next release version

pipiche38, 2019-03-07:
Just check your PR. Is there a way to get also the Cluster on which the bind is made ?

doudz, 2019-07-18:
Doc not up to date

pipiche38, 2019-08-21:
For me the 0x8030 (and true also for 0x8031) are incomplete, as we also need to know what is the cluster for this bind

ISO-B, 2019-08-22:
Correct response for both commands is:
```
<Sequence number: uint8_t>
<Status: uint8_t>
<SrcEndpoint: uint8_t>
<SrcAddrMode: uint8_t>
<SrcAddr: uint16_t>
<ClusterId: uint8_t>
```

pipiche38, 2019-08-22:
Please do check what is really happening. 
The data structure is probably correct. I' not discussing it, but the ClusterId value is set to the ShortAddress !

And SrcAddrMode Type should be based on SrcAddrMode

pipiche38, 2019-11-27:
Will be fixed in 3.1b

<Sequence number: uint8_t>
<Status: uint8_t>
<SrcAddrMode: uint8_t>
<SrcAddr: uint16_t>

closed 2019-11-27 by pipiche38

doudz, 2020-01-27:
This issue should be reopened because cluster ID is not available on 3.1b and 3.1c
see https://github.com/doudz/zigate/issues/170