fairecasoimeme/ZiGate #124: New command 0x0016 - SetTime

pipiche38, 2019-01-16:
I'm struggling a bit on what is written in the zigate.fr as regards to the TimeSet.

From Zigate.fr, the Set Time should be related to <timestamp UTC: uint32_t> from 2001-01-01 00:00:00

which seems to contradict what is written in NXP https://www.nxp.com/docs/en/user-guide/JN-UG-3115.pdf
and which refer to number of seconds since 2000 !

> The Time cluster contains an attribute for the current time, as well as associated 
> information such as time-zone and daylight saving - see Section 18.3
> . The time attribute is referenced to UTC (Co-ordinated Universal Time) an
> d based on the type UTCTime
> , which is defined in the ZigBee standard as:
> "UTCTime is an unsigned 32 bit value representing the number of seconds since 0 
> hours, 0 minutes, 0 seconds, on the 1st of January, 2000 UTC".

pipiche38, 2019-01-16:
So is 2001 or instead 2001 as the ZigBee standard ?

doudz, 2019-01-16:
> So is 2001 or instead 2001 as the ZigBee standard ?

you mean 200**0** or 200**1** ;-)

fairecasoimeme, 2019-01-16:
Oups, you're right, i'll update. 
This is 2000-01-01 00:00:00

closed 2019-02-10 by pipiche38