fairecasoimeme/ZiGate #125: Is the cluster 0x0102 (Window covering) supported by zigate ?

doudz, 2019-01-17:
I can't find any information in NXP doc about the cluster 0x0102, is supported ? and manageable ?

I saw informations about door lock, cluster 0x0101 but nothing about 0x0102

https://github.com/doudz/zigate/issues/63

doudz, 2019-01-17:
https://www.nxp.com/docs/en/user-guide/JN-UG-3076.pdf Page 21

> Shade and window covering solutions are not currently supported by NXP’s ZigBee
> HA profile.

???? really ???

dcolens, 2019-01-17:
I'm curious too could we get support for that cluster ? Here's a device that exposes that uses that cluster:  https://www.legrand.fr/pro/catalogue/42566-version-celiane-with-netatmo/interrupteur-filaire-connecte-pour-volet-roulant-celiane-with-netatmo-blanc (available at Leroy Merlin)

doudz, 2019-01-17:
for information
https://github.com/zigpy/zigpy/blob/master/zigpy/zcl/clusters/closures.py#L133

pipiche38, 2019-01-17:
May be I'm wrong, but the fact that NXP doesn't support could just be that they didn't implement any API for it, while you can still Read/Write Atrributes for that Cluster.

pipiche38, 2019-01-17:
> yes read/write should be possible, but what about command (open,
> close, ...) ?
> I think we can send command using raw aps data request

You have a valid point

ISO-B, 2019-01-17:
It could be possible to write new cluster for windows blinds or generally anything that is missing. It will require quite lot of coding but I think that it would be possible. Hardest part is to figure command response structures. One option is to ask if NXP have plans for implementing it anytime soon.

ISO-B, 2019-01-19:
Found cluster definition for this from ZigBee Cluster Library Specification and it doesn't look that bad. I can try to make this cluster to happen. Does someone have ability to test it if I start working on this?

ISO-B, 2019-01-20:
Managed to make cluster and build binaries. Next thing to do is add user commands and try to test it.
@dcolens Are you interested to test this cluster when I have working binaries?

fairecasoimeme, 2019-01-20:
Hi, I ordered the device. I think i'll be able to test next week.
Thanks

ruimarinho, 2019-01-20:
Would that be the same cluster used by the Aqara Curtain motor? https://m.aliexpress.com/item/32808543920.html

ISO-B, 2019-01-21:
It seems that Aqara Curtain motor uses same 0x0102 Window Covering Cluster. 
Based on: https://github.com/Koenkk/zigbee2mqtt/issues/294 and https://github.com/TedTolboom/com.xiaomi-mi-zigbee/blob/master/drivers/dec to hexurtain/device.js

dcolens, 2019-01-21:
@ISO-B yes, I can test your builds (I use a RPI-3)

ISO-B, 2019-01-21:
Binary can be found: https://github.com/ISO-B/ZiGate-firmware/releases/tag/v3.0f-windowcovering
For me it seems that commands work but I get can't be sure since I don't have anything to test it.

Command: 0x00FA
Message format:
```
<address mode: uint8_t>
<target short address: uint16_t>
<source endpoint: uint8_t>
<destination endpoint: uint8_t>
<cluster command: uint8_t>
<extra parameter: check table below>
```

Cluster command | Description | Extra parameter | Parameter type
------------------|-------------|------------------|-----------------
0x00 | Up/Open | - | -
0x01 | Down/Close | - | -
0x02 | Stop | - | -
0x04 | Go To Lift Value | Value in cm | uint16_t
0x05 | Go To Lift Percentage | Percentage 0-100 | uint8_t 
0x07 | Go To Tilt Value | Value in cm | uint16_t
0x08 | Go To Tilt Percentage | Percentage 0-100 | uint8_t 

Values need to be between destination device's Installed open and closed limits. Sending value outside of those values should make device response status be INVALID_VALUE(Status code 0x25). Same thing with sending percentage value over 100.
Sending LIFT or TILT command to device that doesn't support that should return UNSUPPORTED_COMMAND

dcolens, 2019-01-21:
I loaded the firmware and am trying to use https://github.com/doudz/zigate to send a command to the switch, I tried the following: 
```python3
>>> z.devices[0].endpoints
{1: {'clusters': {0: Cluster 0x0000 General: Basic}, 'out_clusters': [0, 64513, 25], 'profile': 260, 'in_clusters': [0, 3, 258, 4, 5, 15, 64513], 'device': 514}}
>>> z.devices[0].addr     
'082a'
>>> data = struct.pack('!BHBBB', 2, int('082a', 16), 0x01, 1, 0x01) 
>>> z.send_data(0x00FA, data)
WARNING:zigate:No response after command 0x00fa (4)
```
I guess I'm doing something wrong, I bet the 0x00FA is the wrong part because I don't know what it matches to ?

ISO-B, 2019-01-21:
No, it should be 0x00FA. I will look into it tomorrow. Did you managed to pair switch with ZiGate?

doudz, 2019-01-21:
And did you erase eeprom of your zigate ?

dcolens, 2019-01-21:
yes, I erased eeprom but I did not try re-pairing the switch, I just did, but somehow the pairing fails now, the blue led on the zigate goes in fast blink mode and it is not responding to commands anymore.

ISO-B, 2019-01-21:
Rebuilt binary without debug options. Now it should work,

dcolens, 2019-01-22:
@ISO-B I was not able to pair with the latest image either.

ISO-B, 2019-01-22:
I just downloaded it and managed pair couple of devices. Did you do Complete EEPROM erase after flashing it?

pipiche38, 2019-01-22:
@ISO-B while looking after the command why Off is 0x01 and On is 0x00. Shouldn't we use 0x00 for OFF and 0x01 to be consistent with the 0x0092 Command ?

doudz, 2019-01-22:
I include the function `action_cover` in the last 0.26.0

ISO-B, 2019-01-22:
Commands are from ZiGate specs and I think we should use those straight. When we have verified that commands works it's possible to make change commands if wanted. Or just address that in library.

dcolens, 2019-01-22:
@ISO-B whenever I flash the zigate with the image using 
```bash
/JennicModuleProgrammer/Build/JennicModuleProgrammer -V 6 -P 115200 -f ./ZiGate_JN5168_COORDINATOR_115200_3.0f-window_covering.bin -s /dev/ttyUSB1 
JennicModuleProgrammer Version: 0.6 (r56124)
Setting baudrate: 115200
Detected Chip: Unknown
MAC Address:   00:15:8D:00:01:B2:2A:54
Opened firmware file size 209084
Module uses Bootloader v2 Header
Erasing:   100%
Writing Program to Flash
Writing:   100%
Verifying Program in Flash
Verifying: 100%
Success
```
after 30 seconds or so the zigate's blue led starts blinking fast and it is not responding anymore. This happens regardless of having a program using the zigate or not. Restarting the zigate triggers the same after around 30s (maybe 60?) it starts blinking and stops responding to commands.

JennicModuleProgrammer does not list an option to explicitly erase eeprom, I assume it is done by default. If I flash the 3.0e using the same method it does not show the same behavior.

ISO-B, 2019-01-22:
I have only used windows for programming Zigate. I can try tomorrow with Linux and try to resolve how to erase eeprom with jennicmoduleprogrammer. Does anyone else have experience programming with Linux?

lboue, 2019-01-22:
I laso have the same issue, I can't erase EEPROM with Linux: see https://github.com/fairecasoimeme/ZiGate/issues/120

dcolens, 2019-01-23:
Some update, I think I figured how to clean eeprom, I run  
``` 
z.erase_persistent()
z.factory_reset()
```
this seems to do the trick! (the zigate-flasher fails for me https://github.com/tjikkun/zigate-flasher/issues/3)

So now, when I enable permit_join, it seems the zigate fails to pair the legrand shutter (tried 4-5 times to reset the legrand):
```
z.permit_join()

DEBUG:zigate:REQUEST : 0x0049 FFFC1E00
DEBUG:zigate:Msg to send b'0049000450fffc1e00'
DEBUG:zigate:Encoded Msg to send b'010210490210021450fffc1e021003'
DEBUG:zigate:Waiting for status message for command 0x0049
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\x18\x02\x10\xd4\x02\x10I\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'00d40049'
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:212, packet_type:73, error:b'', rssi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0049:RESPONSE 0x8000 - Status response : status:0, sequence:212, packet_type:73, error:b'', rssi:0

<zigate.responses.R8000 at 0x6c433470>

DEBUG:zigate:Auto saving /home/pi/.zigate.json
DEBUG:zigate:Saving persistent file
DEBUG:zigate:File already existing, make a backup before
DEBUG:zigate:Raw packet received, b'\x01\x02\x10M\x02\x10\x02\x1c\xe6\x17\xfb\x02\x10\x02\x14t\x02\x10\x02\x10\x81T`\x8e\x02\x10\x03'
DEBUG:zigate:Raw packet received, b'\x01\x02\x10M\x02\x10\x02\x1c\xad\x17\xfb\x02\x10\x02\x14t\x02\x10\x02\x10\x81T`\x8eK\x03\x01\x87\x02\x11\x02\x10\x02\x13\x85\x02\x10\x02\x10\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x004d: b'17fb00047400008154608e'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:RESPONSE 0x004D - Device announce : addr:17fb, ieee:0004740000815460, mac_capability:10001110, rssi:0
DEBUG:zigate:Device Announce RESPONSE 0x004D - Device announce : addr:17fb, ieee:0004740000815460, mac_capability:10001110, rssi:0
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x004d: b'17fb00047400008154608e'
DEBUG:zigate:RESPONSE 0x004D - Device announce : addr:17fb, ieee:0004740000815460, mac_capability:10001110, rssi:75
DEBUG:zigate:Device Announce RESPONSE 0x004D - Device announce : addr:17fb, ieee:0004740000815460, mac_capability:10001110, rssi:75
DEBUG:zigate:Dispatch ZIGATE_DEVICE_ADDED
DEBUG:zigate:REQUEST : 0x0045 17fb
DEBUG:zigate:Msg to send b'00450002ab17fb'
DEBUG:zigate:Encoded Msg to send b'0102104502100212ab17fb03'
DEBUG:zigate:Waiting for status message for command 0x0045
DEBUG:zigate:Received response 0x8701: b'0000'
DEBUG:zigate:Dispatch ZIGATE_DEVICE_UPDATED
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:RESPONSE 0x8701 - Route Discovery Confirmation : status:0, network_status:0, rssi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\x15\x02\x10\xd5\x02\x10E\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'00d50045'
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:213, packet_type:69, error:b'', rssi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0045:RESPONSE 0x8000 - Status response : status:0, sequence:213, packet_type:69, error:b'', rssi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:Raw packet received, b'\x01\x80E\x02\x10\x02\x17\xb0\xd5\x02\x10\x17\xfb\x02\x11\x02\x11K\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8045: b'd50017fb0101'
DEBUG:zigate:RESPONSE 0x8045 - Active endpoints : endpoints:[OrderedDict([('endpoint', 1)])], sequence:213, status:0, addr:17fb, endpoint_count:1, rssi:75
DEBUG:zigate:REQUEST : 0x0043 b'\x17\xfb\x01'
DEBUG:zigate:Msg to send b'00430003ad17fb01'
DEBUG:zigate:Encoded Msg to send b'0102104302100213ad17fb021103'
DEBUG:zigate:Waiting for status message for command 0x0043
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\x10\x02\x10\xd6\x02\x10C\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'00d60043'
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:214, packet_type:67, error:b'', rssi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0043:RESPONSE 0x8000 - Status response : status:0, sequence:214, packet_type:67, error:b'', rssi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
[... 75 lines cut ...]
DEBUG:zigate:Msg to send b'01200017e20217fb0101010200000000010020000700010e10000000'
DEBUG:zigate:Encoded Msg to send b'01021120021017e2021217fb0211021102110212021002100210021002110210200210021702100211021e1002100210021003'
DEBUG:zigate:Waiting for status message for command 0x0120
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\xaa\x02\x10\x02\x1e\x02\x11 \x02\x10\x03'
DEBUG:zigate:Raw packet received, b'\x01\x81 \x02\x10\x02\x18\x8e\x02\x1e\x17\xfb\x02\x11\x02\x11\x02\x12\x8cK\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'000e0120'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:14, packet_type:288, error:b'', rssi:0
DEBUG:zigate:Received response 0x8120: b'0e17fb0101028c'
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
[... cut ...]
DEBUG:zigate:Dispatch ZIGATE_ATTRIBUTE_ADDED
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:Raw packet received, b'\x01\x80H\x02\x10\x02\x1a\x02\x17\x02\x10\x02\x14t\x02\x10\x02\x10\x81T`\x02\x10\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8048: b'000474000081546000'
DEBUG:zigate:RESPONSE 0x8048 - Leave indication : ieee:0004740000815460, rejoin_status:0, rssi:0
DEBUG:zigate:Dispatch ZIGATE_DEVICE_REMOVED
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
```

doudz, 2019-01-23:
according to log, the device is correctly paired but after it leaves...

ISO-B, 2019-01-23:
Is behavior same if you use `ZiGate_3.0f_pre_release.bin`?

dcolens, 2019-01-24:
@ISO-B yes, I have the same output with `ZiGate_3.0f_pre_release.bin`

fairecasoimeme, 2019-01-24:
I received mine. The procedure to pair Legrand device is the same with all. 
First, put the ZiGate in "permit join" mode 
then you have to powering and when you receive "device announce" (0x004d), you have to disconnect from power. (before leave command)
After you can power a second time and that's ok
here is the clusters 
Type: 0x8043 (Simple Descriptor Response)
  SQN: 0xE8
  Status: 0x00
  Short Address: 0x55BE
  Length: 28
  EndPoint: 0x01
  Profile ID: 0x0104 (ZigBee HA)
  Device ID: 0x0202 (Unknown)
  Input Cluster Count: 7
    Cluster 0:  Cluster ID: 0x0000 (General: Basic)
    Cluster 1:  Cluster ID: 0x0003 (General: Identify)
    Cluster 2:  Cluster ID: 0x0102 (Unknown)
    Cluster 3:  Cluster ID: 0x0004 (General: Groups)
    Cluster 4:  Cluster ID: 0x0005 (General: Scenes)
    Cluster 5:  Cluster ID: 0x000F (General: Binary Input Basic)
    Cluster 6:  Cluster ID: 0xFC01 (Unknown)
  Output Cluster Count: 3
    Cluster 0:  Cluster ID: 0x0000 (General: Basic)
    Cluster 1:  Cluster ID: 0xFC01 (Unknown)
    Cluster 2:  Cluster ID: 0x0019 (General: OTA)

@ISO-B could you send to me your bin to test ?

ISO-B, 2019-01-24:
You should be able to download it from https://github.com/ISO-B/ZiGate-firmware/releases/tag/v3.0f-windowcovering

lboue, 2019-01-24:
You should download this archive for TI chips: [RTOS/CC2652R: optimized z-stack 1.60](https://e2e.ti.com/cfs-file/__key/communityserver-discussions-components-files/158/simplelink_5F00_zigbee_5F00_sdk_5F00_plugin_5F00_1_5F00_60_5F00_00_5F00_14.7z) 
and check zcl_closures.c and zcl_closures.h files:
```
#ifdef ZCL_WINDOWCOVERING
/**********************************************/
/*** Window Covering Cluster Attribute Sets ***/
/**********************************************/
#define ATTRSET_WINDOW_COVERING_INFO                        0x0000
#define ATTRSET_WINDOW_COVERING_SETTINGS                    0x0010

/******************************************/
/*** Window Covering Cluster Attributes ***/
/******************************************/
//Window Covering Information
#define ATTRID_CLOSURES_WINDOW_COVERING_TYPE                ( ATTRSET_WINDOW_COVERING_INFO + 0x0000 )
#define ATTRID_CLOSURES_PHYSICAL_CLOSE_LIMIT_LIFT_CM        ( ATTRSET_WINDOW_COVERING_INFO + 0x0001 )
#define ATTRID_CLOSURES_PHYSICAL_CLOSE_LIMIT_TILT_DDEGREE   ( ATTRSET_WINDOW_COVERING_INFO + 0x0002 )
#define ATTRID_CLOSURES_CURRENT_POSITION_LIFT_CM            ( ATTRSET_WINDOW_COVERING_INFO + 0x0003 )
#define ATTRID_CLOSURES_CURRENT_POSITION_TILT_DDEGREE       ( ATTRSET_WINDOW_COVERING_INFO + 0x0004 )
#define ATTRID_CLOSURES_NUM_OF_ACTUATION_LIFT               ( ATTRSET_WINDOW_COVERING_INFO + 0x0005 )
#define ATTRID_CLOSURES_NUM_OF_ACTUATION_TILT               ( ATTRSET_WINDOW_COVERING_INFO + 0x0006 )
#define ATTRID_CLOSURES_CONFIG_STATUS                       ( ATTRSET_WINDOW_COVERING_INFO + 0x0007 )
#define ATTRID_CLOSURES_CURRENT_POSITION_LIFT_PERCENTAGE    ( ATTRSET_WINDOW_COVERING_INFO + 0x0008 )
#define ATTRID_CLOSURES_CURRENT_POSITION_TILT_PERCENTAGE    ( ATTRSET_WINDOW_COVERING_INFO + 0x0009 )

//Window Covering Setting
#define ATTRID_CLOSURES_INSTALLED_OPEN_LIMIT_LIFT_CM        ( ATTRSET_WINDOW_COVERING_SETTINGS + 0x0000 )
#define ATTRID_CLOSURES_INSTALLED_CLOSED_LIMIT_LIFT_CM      ( ATTRSET_WINDOW_COVERING_SETTINGS + 0x0001 )
#define ATTRID_CLOSURES_INSTALLED_OPEN_LIMIT_TILT_DDEGREE   ( ATTRSET_WINDOW_COVERING_SETTINGS + 0x0002 )
#define ATTRID_CLOSURES_INSTALLED_CLOSED_LIMIT_TILT_DDEGREE ( ATTRSET_WINDOW_COVERING_SETTINGS + 0x0003 )
#define ATTRID_CLOSURES_VELOCITY_LIFT                       ( ATTRSET_WINDOW_COVERING_SETTINGS + 0x0004 )
#define ATTRID_CLOSURES_ACCELERATION_TIME_LIFT              ( ATTRSET_WINDOW_COVERING_SETTINGS + 0x0005 )
#define ATTRID_CLOSURES_DECELERATION_TIME_LIFT              ( ATTRSET_WINDOW_COVERING_SETTINGS + 0x0006 )
#define ATTRID_CLOSURES_WINDOW_COVERING_MODE                ( ATTRSET_WINDOW_COVERING_SETTINGS + 0x0007 )
#define ATTRID_CLOSURES_INTERMEDIATE_SETPOINTS_LIFT         ( ATTRSET_WINDOW_COVERING_SETTINGS + 0x0008 )
#define ATTRID_CLOSURES_INTERMEDIATE_SETPOINTS_TILT         ( ATTRSET_WINDOW_COVERING_SETTINGS + 0x0009 )

/*** Window Covering Type Attribute types ***/
#define CLOSURES_WINDOW_COVERING_TYPE_ROLLERSHADE                       0x00
#define CLOSURES_WINDOW_COVERING_TYPE_ROLLERSHADE_2_MOTOR               0x01
#define CLOSURES_WINDOW_COVERING_TYPE_ROLLERSHADE_EXTERIOR              0x02
#define CLOSURES_WINDOW_COVERING_TYPE_ROLLERSHADE_EXTERIOR_2_MOTOR      0x03
#define CLOSURES_WINDOW_COVERING_TYPE_DRAPERY                           0x04
#define CLOSURES_WINDOW_COVERING_TYPE_AWNING                            0x05
#define CLOSURES_WINDOW_COVERING_TYPE_SHUTTER                           0x06
#define CLOSURES_WINDOW_COVERING_TYPE_TILT_BLIND_TILT_ONLY              0x07
#define CLOSURES_WINDOW_COVERING_TYPE_TILT_BLIND_LIFT_AND_TILT          0x08
#define CLOSURES_WINDOW_COVERING_TYPE_PROJECTOR_SCREEN                  0x09

/****************************************/
/*** Window Covering Cluster Commands ***/
/****************************************/
#define COMMAND_CLOSURES_UP_OPEN                            ( 0x00 )
#define COMMAND_CLOSURES_DOWN_CLOSE                         ( 0x01 )
#define COMMAND_CLOSURES_STOP                               ( 0x02 )
#define COMMAND_CLOSURES_GO_TO_LIFT_VALUE                   ( 0x04 )
#define COMMAND_CLOSURES_GO_TO_LIFT_PERCENTAGE              ( 0x05 )
#define COMMAND_CLOSURES_GO_TO_TILT_VALUE                   ( 0x07 )
#define COMMAND_CLOSURES_GO_TO_TILT_PERCENTAGE              ( 0x08 )

#define ZCL_WC_GOTOVALUEREQ_PAYLOADLEN                      ( 2 )
#define ZCL_WC_GOTOPERCENTAGEREQ_PAYLOADLEN                 ( 1 )

#endif // ZCL_WINDOWCOVERING
```

ISO-B, 2019-01-25:
@fairecasoimeme Did you had time to test cluster? Pushed files to git, but will wait before making PR until someone can confirm that cluster really works.

@lboue Those are same things that can be also found from [Zigbee Cluster Library Specification](http://www.zigbee.org/~zigbeeor/wp-content/uploads/2014/10/07-5123-06-zigbee-cluster-library-specification.pdf). We already have those implemented. Only thing that is left is to test that our cluster implementation sends commands correctly.

dcolens, 2019-01-25:
@fairecasoimeme thanks for the info on pairing procedure, the legrand device is not in the same room as my RPI which is not in the same room as the fuse box (required to power cycle the legrand), I'm afraid I won't be able to follow it :-(

dcolens, 2019-01-25:
Thanks to @fairecasoimeme above, I thought it should be possible to pair just by resetting which should do a reboot (same as power cycle), the following worked:

1. put zigate in `permit_join`
2. reset legrand
3. wait 5-6s and reset legrand
4. wait a few seconds and reset legrand => green led = paired!

It would be great if we could pair after a reset, I don't know why it fails on pairing at 1st try ?

The good news is I can now control my veranda shutter! I just closed and opened it using @ISO-B's firmware and @doudz's latest python zigate :-)

I tried the following 2 commands with success:

- z.action_cover('828f',1,0X01): close
- z.action_cover('828f',1,0X02): stop
- z.action_cover('828f',1,0X00): open

Thanks to all for the time spent on studying the zigbee spec and patching the project, it is great seeing such enthusiasm and responsiveness 👍

ISO-B, 2019-01-25:
Can you also test those other commands? if your device supports those.

dcolens, 2019-01-25:
@ISO-B I tried the other commands (lift and tilt), nothing happened, I'm not sure those are supported, given the switch would not know the size of the shutter ?

Here are the debugs for 0x04:
```python3
z.action_cover('828f', 1, 0x04, 100)

DEBUG:zigate:REQUEST : 0x00fa b'\x02\x82\x8f\x01\x01\x04\x00d'
DEBUG:zigate:Msg to send b'00fa00089d02828f0101040064'
DEBUG:zigate:Encoded Msg to send b'010210fa021002189d0212828f02110211021402106403'
DEBUG:zigate:Waiting for status message for command 0x00fa
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\x7f\x02\x10\x02\x10\x02\x10\xfa\x02\x10\x03\x01\x81\x02\x11\x02\x10\x02\x17Z\x02\x10\x02\x11\x02\x11\x02\x12\x02\x14\x81Z\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'000000fa'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:250, error:b'', rssi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x00fa:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:250, error:b'', rssi:0
DEBUG:zigate:Received response 0x8101: b'000101020481'
DEBUG:zigate:RESPONSE 0x8101 - Default device response : sequence:0, endpoint:1, cluster:258, cmd:4, status:129, rssi:90

<zigate.responses.R8000 at 0x6e2ab970>

DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
```

doudz, 2019-01-25:
status=129 => 0x81

UNSUP_CLUSTER_COMMAND | 0x81 | The specified cluster command is not supported on the device. Command not carried out.
-- | -- | --

ISO-B, 2019-01-25:
Did you also tested GoToPercentage commands?

According to Zigbee Cluster Library Specification page 7-54:
> The device must support either the Go To Lift Percentage or the Go To Tilt Percentage command.

z.action_cover('828f', 1, 0x05, 50)  Lift
z.action_cover('828f', 1, 0x08, 50)  Tilt
One of those commands should move curtain to 50% position

Also what attributes you can request from this device? Valid attribute id's for cluster are 0x0000 - 0x0009 and 0x0010 - 0x0019.

doudz, 2019-01-25:
I don't think lift or tilt (percent or cm) are supported on that device since there no feeback from the controlled motor, the real cover position is unknown

fairecasoimeme, 2019-01-26:
Ok,
sorry for delay but I have updated TestGUI program to test.
I confirm it's ok for Legrand, that's work good only for commands 0, 1 and 2 but as said @doudz, it's normal. Other command are unknow due to hardware.
Thanks for all.

fairecasoimeme, 2019-01-26:
@dcolens  : indeed, reset is enough normally but I didn't find yet reset button. could you help me to find this or there is a trick ...

fairecasoimeme, 2019-01-28:
Thanks but the back of my device is not like this.
[image]

I think device is waiting for special datas from Legrand gateway. I tried to resolve this issue with ZiGate as a Legrand gateway clone but it's not really a success and it's a little bit restrictive.

Smanar, 2019-01-28:
The reset button is on the front side (the other side), but take care, fragile

dcolens, 2019-01-29:
@fairecasoimeme you need to remove the front cover of the switch to see the reset button which is next to the led.

fairecasoimeme, 2019-01-30:
Yep thanks, It was not clear for me but It's OK now... :)

pipiche38, 2019-03-03:
@fairecasoimeme when you did the test of the Legrand shutter. Did you upgarde the firmware of the legrand part before or was it the product unboxed ?

We are currently facing issue to make the Lift part working. 

On/Off works

But Lift % not !

doudz, 2019-03-03:
If it's the same Legrand Switch, the lift commands aren't supported mainly because there's no feedback from the motor to the switch

pipiche38, 2019-03-04:
@doudz Merci, I didn't go through the all thread.

ISO-B, 2019-03-22:
Window covering cluster is implemented and working. If any problems raises about this cluster open new issue.

closed 2019-03-22 by ISO-B

pipiche38, 2019-08-28:
@ISO-B , Is ON equivalent to 100% and OFF to 0% ?

dmsherazi, 2022-09-22:
=

> t I have updated TestGUI program to test.

@fairecasoimeme  Can you let me know where to test in TEST GUI , I failed to find the place where I can test the covering comands in ZGWUI Tool

dmsherazi, 2022-09-25:
When running the debug exe I get this error while opening the port 
[image]

dmsherazi, 2022-09-25:
Solved by pulling the SQLite package in NUGET and rebuilding the ZGWUI again in VS.

[image]