fairecasoimeme/ZiGate #127: Lecture/Ecritue Attributs

Mukatai, 2019-01-17:
Bonjour,

Lorsque je tente de lire/écrire les attributs d'un cluster j'y parviens uniquement pendant quelque secondes apres l'appairage du device. Apres j'obtient tous le temps :

Type: 0x8702 (APS Data Confirm Fail)
  Status: 0xF0
  Source Endpoint: 0x01
  Destination Endpoint: 0x01
  Destination Mode: 0x02
  Destination Address: 0x37A6
  SQN: 0x00

Je voulais savoir si cela était normal ? Le problème se pose surtout avec le capteur de vibration, car pour modifier sa sensibilité je dois écrire un attribut et j'y parviens uniquement pendant quelques seconde apres sont appairage.

J'ai effectué mes testes uniquement sur des devices Xiaomi. D'ailleurs sur ces devices je ne lis pas beaucoup d'attributs, seulement ceux du cluster Basic et ceux que le device me renvoi lors d'un changement d'état. Pour les autres soit pas de réponse soit l'attribut n'est pas valide. (Attribute Type : 0xFF) Avez-vous les memes comportements ?

Je me posais une autre question sur les clusters, quel est la différence entre les Input et Output ?

Merci.

pipiche38, 2019-01-17:
Le capteur de vibration est sur batterie et passe en mode receive-when-idle off. Donc tu ne peux plus y avoir accès après la période d’appairage

Mukatai, 2019-01-18:
Bonjour,

Merci pour vos réponses. Donc je comprend mieux pourquoi je ne pouvais plus les interroger par la suite.

Pour le capteur de vibration, j'avais essayé de cette manière et ca ne fonctionnait pas. J'ai de nouveau essayé et ca fonctionne si j'appui juste avant d'envoyer la commande. Je reçois bien un message inconnu du capteur. Par contre j'ai toujours un "APS Data Confirm Fail" qui apparait après mais bon pas très ennuyant.

Sinon j'ai un autre problème avec la lecture des attributs sur une lampe philips Hue White 806. J'ai un status error lorsque demande n'importe quel attribut :

[image]

Réponse :
Type: 0x8000 (Status)
  Length: 5
  Status: 0x06 (ZigBee Error Code)
  SQN: 0x00
  Message: 

Avez-vous une idée du pourquoi j'obtient ce résultat ? Sur internet je vois que d'autres utilisateurs arrive à récupérer les attributs de cette lampe…

Merci.

fairecasoimeme, 2019-01-18:
Attention le source endpoint doit être 01 (celui de la ZiGate)

Mukatai, 2019-01-21:
Bonjour,

Ok, je comprend mieux pourquoi ca ne marche pas et pourquoi je n'ai pas de réponse sur certains attributs. Tous les exemples que je voyais travaillais sur le 01 donc j'ai pensé que le endpoint source = endpoint destination.

Merci.
@+

closed 2019-03-07 by ISO-B