fairecasoimeme/ZiGate #129: [REQUEST] ZiGate radio library to support for the zigpy library used by ZHA component in Home Assistant

Hedda, 2019-01-21:
Please consider writing a ZiGate radio library that works with the zigpy library for Home Assistant:

https://github.com/zigpy/zigpy

zigpy is a is Zigbee protocol stack integration project that is designed to work with separate radio libraries which can each interface with multiple USB and GPIO radio adapters/modules over different native serial protocols.

This zigpy library is implemented as a native Zigbee Home Automation Hub (ZHA) component for Home Assistant.

https://www.home-assistant.io/components/zha/

The main benefit with the modular design of zigpy is that software like Home Assistant only need to support this one solution as zigpy then abstracts support for multiple radios serial protocols via radio module libraries:

https://github.com/zigpy/zigpy-xbee
https://github.com/zigpy/zigpy-deconz
https://github.com/zigpy/bellows

doudz, 2019-01-21:
I've started to implement it https://github.com/doudz/zigpy-zigate
it's not working yet, I'm waiting for stuff to make test

max5962, 2019-01-23:
what will the pros and the cons of using zigpy vs your component @doudz  ?

doudz, 2019-01-24:
Pros :
- well integrated in HA
- Better zigbee protocol support
Cons :
- low level so it doesn't use the zigate commands
- not compatible with python 3.4 (not a problem for HA but I can't reuse it for Jeedom for example)

Hedda, 2019-01-24:
Another pro for end-users is zigpy's open modular design made to support different Zigbee radio modules (radios from different hardware manufacturers that does not use the same low-level protocol to communicate with the hardware). Because of zigpy's modular design, at least in theory, you will get more developers with different Zigbee adapter hardware working on the same interface integration and open source libraries for that component.

From an end-users point-of-view it should in theory not matter which Zigbee adapter you buy, as long as some developer(s) have written a radio library for zigbee for the low-level protocol that it is using.

This modular design is very similar to that which have already existed for Z-Wave in the open source community for many years, where you can buy almost USB or GPIO Z-Wave controller adapter from any hardware manufacturer and it will "just work" with all open source home automation software since they use the same library (ie. OpenZWave).

Sure some Zigbee adapter hardware manufacturers who do not yet get the open source concept might not like the modular design of zigbee since it gives end-users the option to go buy hardware adapters from their competition, but I think that this modular design will benefit everyone, including Zigbee adapter hardware manufacturers, developers, and end-users.

max5962, 2019-01-24:
Hello.
First thanks you for  detailed answer!

@doudz what's the issues of not using zigate commands ?

doudz, 2019-01-25:
@max5962 zigpy generate raw zigbee command packet so we will only use a few set of zigate commands like permit join or raw aps data request, the others are just unneeded

Hedda, 2019-01-25:
@doudz would it be any point for @fairecasoimeme to further extend the serial protocol / serial API of the ZiGate firmware so that zigpy could make more advanced use of its firmware features and function?

Hedda, 2019-03-20:
@doudz congratulations on making such great progress with your zigpy-zigate project for Zigpy!

Hedda, 2019-08-20:
@doudz any news to share to the public about this? I see that you updated your zigpy-zigate repo:

https://github.com/doudz/zigpy-zigate

Also noticed that 3.1a firmware released for ZiGate by fairecasoimeme added raw mode support

doudz, 2019-08-20:
@Hedda zigate support has been added to the next Home Assistant release
https://github.com/home-assistant/home-assistant/pull/25552
So tests can be done using HA dev branch

ruimarinho, 2019-09-14:
@doudz what's the advantage of the version included in Home Assistant?

doudz, 2019-09-15:
@ruimarinho it's still a bit experimental but the zigpy version is well integrated in HA so you could manage device in a better way, adjust device parameter, etc...
But some functions are missing, like led management, firmware upgrade

Theorically, using zigpy we will handle zigbee better

pipiche38, 2020-09-30:
should be track on https://github.com/zigpy/zigpy-zigate

closed 2020-09-30 by fairecasoimeme

Hedda, 2020-09-30:
FYI, there are several follow-up requests listed in https://github.com/fairecasoimeme/ZiGate/issues/319 in regards to requested help from ZiGate community with zigpy-zigate.

Summery; zigpy-zigate radio library API does still not quite have feature parity with the upstream ZiGate API so help is wanted.