fairecasoimeme/ZiGate #133: Réponse 0x8002 ?

doudz, 2019-01-25:
Dans quel cas peut-on avoir la réponse 0x8002 ? E_SL_MSG_DATA_INDICATION
J'ai l'impression que cela permet d'avoir des données brut mais je ne reçois jamais cette réponse
Cela pourrait être intéressant dans le cadre de l'intégration avec zigpy

fairecasoimeme, 2019-01-26:
Je me trompe peut-être mais c'est une commande interne à la stack ZigBee. Ce n'est pas une réponse radio classique

Je pense qu'il faut regarder du côté de la commande E_SL_MSG_SEND_RAW_APS_DATA_PACKET pour envoyer de commandes brutes.

Pour ce qui est de recevoir des données brutes, par contre, il faut creuser mais peut-être qu'il n'y a pas beaucoup à changer dans le firmware...

doudz, 2019-01-26:
Pour E_SL_MSG_SEND_RAW_APS_DATA_PACKET effectivement c'est la bonne commande.
Concernant la réception il faut que je creuse encore un peu mais s'il s'avère que la réception de donnée brut est nécessaire, on pourrait imaginer une option à activer pour permettre à la zigate de passer en "mode données brutes"

pipiche38, 2020-06-29:
#307

pipiche38, 2020-09-30:
implemented in 31d. All messages not processed by the firmware will be reported as 0x8002

closed 2020-09-30 by fairecasoimeme