fairecasoimeme/ZiGate #134: Data type 0x25 not supported ?

doudz, 2019-01-30:
Wrong data size in 8102 response:

Raw response from zigate : 01810212021011c526a3230211021102110215021802102502100218021002100214489c03

Decoded response 8102 : 26 a323 01 0101 0508 00 **25** **0008** 00000448
As you can see, data size is set to 8 but there only 4 bytes
And data type 0x25 is supposed to be a 48bits so 6 bytes

fairecasoimeme, 2019-01-30:
Ok, I'll fix that tomorrow, for the 3.0f

closed 2019-03-07 by ISO-B

doudz, 2019-03-08:
I'm sorry but it's not solved...

2019-03-08 20:27:35 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x81\x02\x12\x02\x10\x11\xc6\xd4\x95\xf8\x02\x11\x02\x11\x02\x11\x02\x15\x02\x18\x02\x10%\x02\x10\x02\x18\x02\x10\x02\x10\x02\x14\xa8`\x03'
2019-03-08 20:27:35 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-03-08 20:27:35 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8102: b'd495f8010101050800250008000004a8'
2019-03-08 20:27:35 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8102 - Individual Attribute Report : sequence:212, addr:95f8, endpoint:1, cluster:257, attribute:1288, status:0, data_type:37, size:8, data:000004a8, lqi:96