fairecasoimeme/ZiGate #138: Support for KSentry KS-SM001 switch

waltervl, 2019-02-04:
I bought a ZigBee switch Ksentry KS-SM001 https://ksentry.manufacturer.globalsources.com/si/6008837134660/pdtl/ZigBee-module/1162731630/zigbee-on-off-controller-modules.htm (not arrived yet).
This switch is supported already in Zigbee2MQTT: https://github.com/Koenkk/zigbee-shepherd-converters/search?q=KS-SM001&unscoped_q=KS-SM001

Is ZiGate also supporting this switch? If not, what do I have to do to help you have it supported?

pipiche38, 2019-02-05:
To follow

waltervl, 2019-02-21:
First impression of another user seems that it works as expected! http://www.domoticz.com/forum/viewtopic.php?f=68&t=26783#p207597

 Mine is still on its way so I cannot confirm

lboue, 2019-02-27:
@waltervl Did you order on globalsources because minumum is 100 pieces?
I found this one [Remote-Control-Smart-Switch](https://www.aliexpress.com/item/New-Arrival-2019-Remote-Control-Smart-Switch-Timing-Energy-Saving-Compatible-with-Smart-Things-Hub-Wink/32957480204.html?spm=a2g0s.13010108.99999999.9.58f5321eUuH2IF) on Aliexpress.

waltervl, 2019-02-27:
I ordered on AliExpress too, this one:
https://s.click.aliexpress.com/e/SfANbnil5
Still on transport. Chines new year did not help.... :-)

waltervl, 2019-03-05:
They arrived today. Worked directly when added to Zigate in Domoticz.
Domoticz Zigate plugin log when attached (pre-4.1):
```
2019-03-05 18:44:45.127  Status: (ZiGate) [-] NEW OBJECT: 55ec Simple Descriptor EP 0b
2019-03-05 18:44:45.127  (ZiGate) Decode8043 - Overwrite ProfileID {} with 0104 from Ep: 0b 
2019-03-05 18:44:45.127  Status: (ZiGate) [-] NEW OBJECT: 55ec ProfileID 0104
2019-03-05 18:44:45.127  (ZiGate) Decode8043 - Overwrite ZDeviceID {} with 0104 from Ep: 0b 
2019-03-05 18:44:45.127  Status: (ZiGate) [-] NEW OBJECT: 55ec ZDeviceID 0100
2019-03-05 18:44:45.127  Status: (ZiGate) [-] NEW OBJECT: 55ec Cluster IN Count: 05
2019-03-05 18:44:45.127  Status: (ZiGate) [-] NEW OBJECT: 55ec Cluster In 1: 0000
2019-03-05 18:44:45.127  Status: (ZiGate) [-] NEW OBJECT: 55ec Cluster In 2: 0003
2019-03-05 18:44:45.127  Status: (ZiGate) [-] NEW OBJECT: 55ec Cluster In 3: 0004
2019-03-05 18:44:45.127  Status: (ZiGate) [-] NEW OBJECT: 55ec Cluster In 4: 0005
2019-03-05 18:44:45.127  Status: (ZiGate) [-] NEW OBJECT: 55ec Cluster In 5: 0006
2019-03-05 18:44:45.127  Status: (ZiGate) [-] NEW OBJECT: 55ec Cluster OUT Count: 01
2019-03-05 18:44:45.127  Status: (ZiGate) [-] NEW OBJECT: 55ec Cluster Out 1: 0000
2019-03-05 18:44:49.960  Status: (ZiGate) [0] NEW OBJECT: 55ec Request Model Name
2019-03-05 18:44:49.961  (ZiGate) Request Basic  via Read Attribute request part1 55ec/0b [0, 1, 2, 3, 4, 5]
2019-03-05 18:44:49.961  (ZiGate) Request Basic  via Read Attribute request part2 55ec/0b [6, 7, 10, 15, 16, 21]
2019-03-05 18:44:49.962  (ZiGate) Too early, let's try to get the Model
2019-03-05 18:44:50.069  (ZiGate) ReadCluster - 0000 / 0005 - Recepion Model: >Lamp_01<
2019-03-05 18:44:52.718  (Nefit Easy) P1 Smart Meter (CV Gasverbruik)
2019-03-05 18:44:54.964  Status: (ZiGate) [1] NEW OBJECT: 55ec Model Name: Lamp_01
2019-03-05 18:44:54.964  Status: (ZiGate) [1] NEW OBJECT: 55ec Model Name: Lamp_01
2019-03-05 18:44:54.964  Status: (ZiGate) [1] NEW OBJECT: 55ec Model Name: Lamp_01
2019-03-05 18:44:54.964  (ZiGate) Too early, let's try to get the Model
2019-03-05 18:45:00.003  Status: (ZiGate) [2] NEW OBJECT: 55ec Model Name: Lamp_01
2019-03-05 18:45:00.003  Status: (ZiGate) [2] NEW OBJECT: 55ec Model Name: Lamp_01
2019-03-05 18:45:00.003  Status: (ZiGate) [2] NEW OBJECT: 55ec Model Name: Lamp_01
2019-03-05 18:45:00.004  (ZiGate) GetType - Model Lamp_01 Profile ID : 0104 ZDeviceID : 0100
2019-03-05 18:45:00.005  (ZiGate) GetType - Model Lamp_01 not found with Ep: 0b in DeviceConf. Continue with ClusterSearch
2019-03-05 18:45:00.005  (ZiGate) GetType - ClusterSearch return : Switch
2019-03-05 18:45:00.009  (ZiGate) Request a Bind for 55ec/0b on Cluster 0006
2019-03-05 18:45:00.010  (ZiGate) Request a Bind for 55ec/0b on Cluster 0000
2019-03-05 18:45:00.011  (ZiGate) Request Basic  via Read Attribute request part1 55ec/0b [0, 1, 2, 3, 4, 5]
2019-03-05 18:45:00.011  (ZiGate) Request Basic  via Read Attribute request part2 55ec/0b [6, 7, 10, 15, 16, 21]
2019-03-05 18:45:00.553  (ZiGate) UpdateDevice - (ZiGate - Switch-01124b001bafeeb7-0b) 1:On
2019-03-05 18:45:00.667  (ZiGate) ReadCluster - 0000 / 0005 - Recepion Model: >Lamp_01<
2019-03-05 18:45:00.667  (ZiGate) ReadCluster - 0000 / 0005 - 55ec Lamp_01 is already provisioned in Domoticz

```
Device file:
55ec : {'Version': '3', 'Status': 'inDB', 'SQN': {}, 'Ep': {'0b': {'0000': {}, '0003': {}, '0004': {}, '0005': {}, '0006': {}, 'ClusterType': {'108': 'Switch'}}}, 'Heartbeat': '3', 'RIA': '3', 'RSSI': 228, 'Battery': {}, 'Model': 'Lamp_01', 'MacCapa': '8e', 'IEEE': '01124b001bafeeb7', 'Type': 'Switch', 'ProfileID': '0104', 'ZDeviceID': '0100', 'Stamp': {'Time': '2019-03-05 18:44:50', 'MsgType': '8102'}, 'NbEp': '1', 'ConfigSource': '8043', 'ReadAttributes': {'Ep': {'0b': {'0000': {'0000': '00', '0001': '00', '0002': '86', '0003': '00', '0004': '00', '0005': '00', '0006': '00', '0007': '00', '000a': '86', '000f': '86', '0010': '00', '0015': '86'}, '0006': {}}}, 'TimeStamps': {'0b-0000': 1551807900, '0b-0006': 1551807900}}, 'ZCL Version': '1', 'App Version': '1', 'HW Version': '1', 'Manufacturer': 'SZ', 'Date Code': '20181107', 'Location': '', 'Bind': {'0006': {'Stamp': 1551807900, 'Phase': 'requested', 'Status': ''}, '0000': {'Stamp': 1551807900, 'Phase': 'requested', 'Status': ''}}, 'ConfigureReporting': {'Ep': {'0b': {'0000': '', '0003': {}, '0004': {}, '0005': {}, '0006': {}}}, 'TimeStamps': {'0b-0000': 1551807900, '0b-0003': 0, '0b-0004': 0, '0b-0005': 0, '0b-0006': 1551807900}}}

pipiche38, 2019-03-05:
thanks for sharing , good news to get one more device compatible with Zigate

ISO-B, 2019-03-06:
Lamp_01 seems to be model name for that device. Name comes from device firmware so there is nothing we can do. It's probably possible to give it better name in Domoticz.  

It seems to work as it should so this can be closed. @fairecasoimeme can you add this to Compatible hardware section?

closed 2019-03-06 by ISO-B

fairecasoimeme, 2019-03-06:
thanks, this remote is partially compatible with v3.0f and fully compatible with the v3.10. I'll add to the compatible page.

pipiche38, 2019-03-06:
What do you mean partially compatible?

fairecasoimeme, 2019-03-06:
There is a lack of informations on some push button

pipiche38, 2019-03-06:
But that is a simple switch, no ? Are you talking about that one, or the Innr remote ?