fairecasoimeme/ZiGate #139: model identifier sur volet Profalux

doudz, 2019-02-09:
Il semblerait que les volets Profalux ne possède pas de model identifier (cluster 0x0000, attribut 0x0005) car sur les retours de différents utilisateurs, cette valeur est vide.
Mais ça me semble vraiment bizarre du coup je me pose la question : est-ce que réellement les profalux n'ont pas de model identifier ? ou est-ce un bug de la zigate qui n'arrive pas à remonter l'info ?

@fairecasoimeme je vois que tu as testé, aurais-tu des infos ?
https://zigate.fr/moteur-profalux-zoe-ha1-2-compatible-zigate-clusters/

pipiche38, 2019-02-09:
Je confirme.

J’ai 7 profalux et je n’ai pas de Model identifier.

doudz, 2019-02-09:
Et as-tu essayer de sniffer pour voir si qqc arrivait en réalité ?

pipiche38, 2019-02-11:
Je regarde dès que je peu (car sur la production), en plus il va falloir que je les re-paires (pour la passage en 3.0f ;-)

pipiche38, 2019-02-13:
@doudz , juste pour te dire que j'y travail, mais n'ayant jamais utilisé de snifferr, je galère un peu. Dès que j'arrive à décoder le payload, je te donne l'information ;-)

KiwiHC16, 2019-02-13:
Le champ est vide.

pipiche38, 2019-02-13:
@doudz Voici ce que j'obtiens : 

La demande ( Read Attribute )
[image]

La réponse
[image]

J'en déduit que le Model n'est pas supporté par le constructeur.

doudz, 2019-02-13:
Merci beaucoup pour ces retours, le code 0x86 indique que l'attribut n'est pas supporté tout simplement.
J'étais persuadé que c'était un attribut obligatoire mais non en fait...
Dommage c'est bien pratique ;)

closed 2019-02-13 by doudz

pipiche38, 2019-02-28:
@doudz , pour info, je viens de trouver sur l'attribut 0x0010 du cluster 0x0000 le nom que j'avais attribué à mes volets via la télécommande Zoe.

Donc dans le cas du Profalux. cet attribut peut etre intéressant et serait donc RW !

doudz, 2019-02-28:
bien vu !
c'est intéressant, à voir comment ça peut être exploité
merci pour l'info