fairecasoimeme/ZiGate #141: Erase PDM

deennoo, 2019-02-10:
Hello,

Since More and More home automation using Zigate are ready to work, ir's a bit hard for user like my who have more than 25 devices to repair them on each FW update.

Clearly i know that update improve stability, and fonctions but reppairing Ikea Bulb isn't easy, and less easy when there are outside home and that a rainy day.

Please can you help user, who are happy to have an easy solutoin to be more happy and proud of their choise ?

pipiche38, 2019-02-10:
+1 to follow

ISO-B, 2019-02-10:
I can take look to this next week.

Way I think this could be done is dumping data from pdm before flashing and someway adding needed data back after flashing.

tjikkun, 2019-02-11:
Well, ideally the version of the stored data is stored, en there are migrations defined, so when the new firmware loads, it checks version number of the stored data and if it doesn't match the version needed by the firmware, a migration is performed.

pipiche38, 2019-02-11:
> 
> 
> I can take look to this next week.
> 
> Way I think this could be done is dumping data from pdm before flashing and someway adding needed data back after flashing.

I suspect that we need to find the API in the NXP Sigbee stack in order to re-inject pairing information.
From a general view I see:
- Extracting the data in a readable format
- Re-injecting using the API 

If there is no API, that might be problematic as the new firmware could have an impact to the data format

pipiche38, 2019-02-11:
#120 is touching the subject, but works only when the Format doesn't change. So in charge does work when we do not have to erase PDM

waltervl, 2019-02-11:
Is the firmware flashing workaround for other popular open source ZigBee controller working here also?
In short: flash the device somewhere were it has no contact with the original Zigbee network (eg drive 1 km away...) . The PANID and chanel hopefully stays the same in this way and the controller is accepted again by the nodes when reentering the Zigbee network.
I am just starting to explore functionality of Zigbee. ZiGate possible has total different architecture here so this workaround won't work.

ISO-B, 2019-02-12:
Results from testing:
Only thing that we read from ZiGate PDM at startup is ZLLstate which keeps track of ZiGate state. If PDM is empty then state is FACTORY_NEW and node state is STARTUP. After ZiGate has started network successfully then state is NOT_FACTORY_NEW and node state is RUNNING. 

When state is STARTUP ZiGate will generate random PANID and set channel. Just resetting same channel and PANID after eeprom reset won't let devices to connect back. Noticed that FlashGui erase pdm only doesn't do anything.

Debugging this is quite hard since there is no .c files for most low level stuff so can't easily figure where data is in PDM and which format. I might have found some useful functions but need to test those to see if they are useful or not.

fairecasoimeme, 2019-02-12:
Normally, This is the last time that it's mandatory to erase PDM. I worked a lot on this issue and now I think that the data structure is optimized and that we will not update this part ever.

I did several tests with backup and restore with different firmware and It's ok.

I hope and I pray for this ;)

ISO-B, 2019-02-24:
Took some time, but finally managed to figure what structures are on which PDM addresses. I was able to build firmware that was able to restore paired devices after flashing and completely erasing EEPROM. 

If we add version number to PDM and keep track of couple structure sizes per version in code it should be possible to automatically migrate devices when changing version if needed. This will need still some work and testing.

doudz, 2019-02-24:
@ISO-B you're amazing ! :)

pipiche38, 2019-02-26:
@ISO-B did you check that the binding table is also kept ?

ISO-B, 2019-02-27:
I didn't have any devices binded, but I found PDM address for binding table so I assume that bindings are also kept.

ISO-B, 2019-03-01:
Finally got it working after almost 150 firmware builds. #166

ISO-B, 2019-03-05:
Next release version installation should be able to handle possible structure size changes.

closed 2019-03-05 by ISO-B