fairecasoimeme/ZiGate #143: After upgrading to 3.0f I can't pair my Hue bulbs again.

santricity, 2019-02-10:
Hi,
I just upgraded firmware to 3.0f, all devices (ikea bulbs, aqara pirs, osram) paired successfully except Philips Hue bulbs. I don't have Philips remote to force them into pairing mode. Is there any way to make them working with Zigate again? I'm running on stable latest Zigate Plugin, Domoticz 4.9700 on up to date Raspbian.

santricity, 2019-02-13:
Should I really buy Hue Remote to make it working again?

doudz, 2019-02-13:
I'm not sure but maybe you could use the commande 0x00D2 "Touch link factory reset target" ? Because I suppose that's the HUE remote is doing
if someone could confirm

santricity, 2019-02-13:
Thank you! Could you let me know how to send this command to the device?

doudz, 2019-02-13:
Don't know using Domoticz, so probably with ZGUi

santricity, 2019-02-13:
Could you let me know how I can send this command through the ZGUI? I installed it, connected the USB ZiGate stick, I can discover some devices, but I don't know where I should put this command. Too many options and fields...

pipiche38, 2019-02-13:
@fairecasoimeme is there any specific condition to do the Factory reset Touch Link ? Is this really reseting the Hub Bulb and make them able to re-pair ?

santricity, 2019-02-18:
Fixed by ordering Hue dimmer remote.

sifus, 2019-02-19:
hi, what is the adress to the 3.0f Bin ?

pipiche38, 2019-02-19:
@sifus : https://github.com/fairecasoimeme/ZiGate/releases

sifus, 2019-02-19:
Sorry, i try to get this bin with linux command :
./JennicModuleProgrammer -V 6 -P 115200 -f « < ??? > » -s /dev/ttyUSB1

pipiche38, 2019-07-14:
I think this can be closed !

closed 2019-07-15 by ISO-B