fairecasoimeme/ZiGate #145: Add TxPower control with 0x0806 command / Get TxPower should be made available too

pipiche38, 2019-02-11:
In regards to set TxPower , it would be convenient to have get TxPower as well

pipiche38, 2019-02-18:
@fairecasoimeme 

Could you clarify what are the expected value for the TXPower command ?

Is that a value between 0 and 0xbf ?
If 0x80 the default value ?

pipiche38, 2019-02-19:
Quel devrait etre le résultat attendu de l'utilisation du TX Power ?

Est-ce les bonnes valeurs ?

pipiche38, 2019-02-19:
BTW, while going to the source code, if I'm not mistaken,
this Command is always return SUCCESS!!! Looks like the return of eAppApiPlmeSet  functions is not taken in consideration. a PHY_Enum_e enumeration status value to indicate success or failure of the operation.

```

        if (1 == u16PacketLength)
        {
                u8TxPower = pu8LinkRxBuffer[ u32BytesRead ];
                u32BytesRead += sizeof(u8TxPower);

                eAppApiPlmeSet(PHY_PIB_ATTR_TX_POWER, u8TxPower);
                *peAHIStatus = E_AHI_SUCCESS;
        }

```

pipiche38, 2019-03-15:
What is the expected impact of this 3.0f parameter ?

ISO-B, 2019-03-15:
I am currently changing command to return something else than always success. Also adding Get TxPower command.

pipiche38, 2019-03-15:
Great. And what about the correct values : 
Is that a value between 0 and 0xbf ?
If 0x80 the default value ?

ISO-B, 2019-03-15:
That seems correct based on NXP IEEE 802.15.4 Stack User Guide (JN-UG-3024 v2.6) Table 10
```
PHY_PIB_TX_POWER_DEF (default - 0x80)
PHY_PIB_TX_POWER_MIN (minimum - 0)
PHY_PIB_TX_POWER_MAX (maximum - 0xbf)
```

ISO-B, 2019-03-27:
This should now work on v3.1a

closed 2019-03-27 by ISO-B

pipiche38, 2019-06-21:
confirmed

aztazt, 2019-08-09:
I am supposed to see a Get TX button in the TestGUI_Zigate software ? Sorry, I'm new to zigbee.
Thank you