fairecasoimeme/ZiGate #147: 0x8085 returning always the same Command despite button press with HUE Remote

pipiche38, 2019-02-12:
Seems that Decode8085 introduced with 3.0f is returning always

0x02/0x02 on cluster 0x0008 when pressing '+' Hue remote command
0x03/0x02 on cluster 0x0008 when pressing '-' Hue remote command

deennoo, 2019-02-12:
Philips Hue Remote pair to an Ikea Bulb give this on ZGWUI

For + (or Dimmer up) :
Type: 0x8085 (Unrecognized)
21:22:19.462 <- 01 80 85 00 09 9E BF 01 00 08 02 C4 6D 02 8D 03

For - (or Dimmer Down) :
Type: 0x8085 (Unrecognized)
21:23:01.070 <- 01 80 85 00 09 E1 C0 01 00 08 02 C4 6D 02 8D 03

pipiche38, 2019-02-13:
21:22:19.462 <- 01 80 85 00 09 9E BF 01 00 08 02 C4 6D 02 8D 03
ClusterID: 00 08 
SrcAddr: C4 6D 
Cmd: 02 
RSSI 8D 

01 80 85 00 09 E1 C0 01 00 08 02 C4 6D 02 8D 03
SrcAddr: C4 6D 
CMD: 02 
RSSI: 8D

Which demonstrate that from ZGUI we get the same value despite + or - is pressed !

@fairecasoimeme , @ISO-B any ideas

ISO-B, 2019-02-13:
I don't have Phillips remote so I can't test this. 
Command 0x02 is for step. In event handling we should look into command payload. There is step mode value in payload which will identify which button is pressed.  In case of command 0x02 we should send this step mode value to user along side cmd value(This is currently sent).

Without implementing 0xFC00 cluster we can only get one event per button. By implement that cluster we could get down, hold, short and long release events from all button.

pipiche38, 2019-02-13:
I was under the impression (from Release notes of 3.0f) that 0xFC00 was implemented, no ?

fairecasoimeme, 2019-02-13:
0xFC00 is implemented in 3.0f.
But I have to explain how exploit this function with bind command.

fairecasoimeme, 2019-02-13:
The good method to intercept hue remote control is : 
* pair the hue remote to ZiGate
* execute bind command with the following parameter

<target extended address: uint64_t> --> philips hue remote IEEE address
<target endpoint: uint8_t> --> 0x02
<cluster ID: uint16_t> 0xFC00
<destination address mode: uint8_t>-->  IEEE mode (i think 0x03)
<destination address:uint64_t> -->  ZiGate IEEE address
<destination endpoint : uint8_t> --> 0x01

ex: 01 02 10 30 02 10 15 02 10 02 10 17 88 02 11 02 12 1A E3 DF 02 12 FC 02 10 02 13 02 10 15 8D 02 10 02 11 B2 2E 66 02 11 03 

After, ZiGate can intercept all of key press with this kind of datas : 

<Attrib ID : uint16_t>
<Attr data type : uint8_t>
<Attr Size : uint16_t>
<Attr datas : string>

example :
 **01** 00 42 00 05 30 00 21 00 00 --> On button
 **04** 00 42 00 05 30 02 21 01 00 --> Off button
 **02** 00 42 00 05 30 00 21 00 00 --> Level up
 **03** 00 42 00 05 30 02 21 01 00 --> Level down

Datas structure : 
<Datatype : uint8_t> --> 30 = enum8
<Action type : uint8_t> --> 0 = short press / 1 = long press / 2 = release (short press) / 3 = release (long press) 
<Data type : uint8_t> --> 21 =  uint16_t
<Time pushed : uint16_t> --> in second

I hope it's more clear... I'll update the doc of course

pipiche38, 2019-02-14:
Ok, so the trick is to use EP "02" instead of standard EP "01" , will test it

pipiche38, 2019-02-14:
As I don't have the Hue remote are those event coming from the 0x8095 and 0x8085

fairecasoimeme, 2019-02-14:
Nop, 0x8102 messages like that :
01 81 02 00 12 E5 00 D1 77 02 FC 00 00 01 00 42 00 05 30 00 21 00 00 7B 03
21

pipiche38, 2019-02-16:
Thanks @fairecasoimeme this is really helpfull.
I close the issue

closed 2019-02-16 by pipiche38