fairecasoimeme/ZiGate #148: Livolo 2 Gang Zigbee Switch

deennoo, 2019-02-12:
Hi there.

I just get a 2 Gang Livolo Zigbee Switch.

This run on channel 26 only (thanx : https://github.com/Koenkk/zigbee2mqtt/issues/592)

Trying to pair it with zigate looks to works but nothing is working

there is some log from ZGWUI : 

premiere connection :

```
Type: 0x004D (End Device Announce)
  Short Address: 0x283D
  Extended Address: 0x124B0018015B39
  MAC Capability: 0x80
    Alternate PAN Coordinator: False
    Device Type: End Device
    Power Source: Battery
    Receiver On When Idle: False
    Security Capability: Standard
    Allocate Address: True
Type: 0x004D (End Device Announce)
  Short Address: 0x283D
  Extended Address: 0x124B0018015B39
  MAC Capability: 0x80
    Alternate PAN Coordinator: False
    Device Type: End Device
    Power Source: Battery
    Receiver On When Idle: False
    Security Capability: Standard
    Allocate Address: True
Type: 0x8701 (Route Discovery Confirm)
  SQN: 0x00
  Status: 0x00
  Network Status: 0x00
```
EP Request :

```
[Type: 0x8000 (Status)
  Length: 5
  Status: 0x00 (Success)
  SQN: 0x24
  Message: 
Type: 0x8045 (Active Endpoints Response)
  SQN: 0x24
  Status: 0x00
  Short Address: 0x283D
  Endpoint Count: 1
  Endpoint List: 
    Endpoint 0: 0x06](url)
```
Cluster Request :

```
EndPoint: 0x06
Profile ID: 0x0104 (ZigBee HA)
Device ID: 0x0000 (Generic - On/Off Switch)
Input Cluster Count: 2
  Cluster 0:  Cluster ID: 0x0000 (General: Basic)
  Cluster 1:  Cluster ID: 0x0003 (General: Identify)
Output Cluster Count: 1
  Cluster 0:  Cluster ID: 0x0006 (General: On/Off)
```
Clearly End Point and Cluster request aren't good as this is a 2 gang device. 

After Pairing, nothing comefrom the device to the zigate when switching on/off each relay.

Device doesn't react from ZGWUI solicitations :

* ON/Off
* Identify

Some help can be good, first to drive it from ZGWUI

KiwiHC16, 2019-02-13:
Mine works very well with ZiGate/Abeille/Jeedom

pipiche38, 2019-02-18:
@deennoo , I suggest that we track this one on the plugin side.

pipiche38, 2019-02-21:
As discussed in parallel, the beacon topic from my end is difficult to be managed from the plugin itself. This is from my point of view something which should be handled by Zigate it self as part of the ZigBee transport.

ISO-B, 2019-03-25:
https://github.com/sasu-drooz/Domoticz-Zigate/issues/337

>
>"There is a difference in the switch behaviour that depends on coordinator. The switch always sends 'Data request' commands 3-4 times per second (!!!). If there is no answer even for less than a second the switch sends 'Orphan Notification' then 'Beacon Requests'. If it is paired with original livolo gateway, it's ok, it sends these requests as long as it's necessary, but if it is custom coordinator the switch stops working after a second, and there is only one way to make it work again, it's touch the sensor for 5 secs or power it off then on. So, my switch works fine as long as the connection is perfect, in other way it stops working after some time."

pipiche38, 2019-04-05:
Found on: https://github.com/Koenkk/zigbee2mqtt/issues/592

extPanId must be 0x00124b00198d7521 if it's not, change your pan_id to force the coordinator to change ext_pan_id.
Also pay attention that permit_join always must be true so it's better to use the dedicated coordinator for livolo.

KiwiHC16, 2019-07-23:
I'm re-investigating this topic. I had a 2 buttons which worked for days and months. I decided to replace it by a one button and I have this syndrome of switch disconnection after couple of minutes... Very strange... Work like a charme a couple of minutes and then ...

pipiche38, 2019-07-23:
For me Livolo seems to be not compliant. When looking after Dresden for instance they recommend to have a dedicated Key to manage a max of 6 Livolo, due to the high level of communication!

fairecasoimeme, 2019-07-25:
I thinks (like the 2 buttons) that it is waiting for some command or informations. The solution is to buy the livolo gateway and sniff. https://fr.aliexpress.com/item/32918061484.html
That's what I did ;)

KiwiHC16, 2019-07-25:
Yes there is a trick that you see if you sniff. 
The issue of such switch, as xiaomi switch, is that they poll the network for instructions. With LIVOLO every 1/3 of a second. Like this if you send a switch command from the GW you have the feeling that it s instantaneous.  Now this request goes to the router of attachment. What is the duration of a data request ? Base on that we could estimate the number of data request which could be sent in 0.3s. So I think that the limit of 10 as no real justification. I don t understand why they don t do like Legrand and implement a router in the switch.....

pipiche38, 2019-10-14:
For info, I found a way to get Livolo been control from Channel 11 ! 
I'll see how stable it is ! But so far so good

It is controlable Switch On and Off

with pre-3.1b the state is corretly provided by the switch

KiwiHC16, 2019-10-29:
What is the trick to change the frequency ?

pipiche38, 2019-11-21:
With firmware 3.1b it is interesting to see that when the Livolo switch is flying from channel 26 to 11, it is doing a Device Annouced with flag Rejoin set to 02

crus1313, 2020-02-11:
> For info, I found a way to get Livolo been control from Channel 11 !
> I'll see how stable it is ! But so far so good
> 
> It is controlable Switch On and Off
> 
> with pre-3.1b the state is corretly provided by the switch

Could you share how you did that? And why is ch 11 better?

pipiche38, 2020-02-11:
Ch 11 is better because quiet a lot of devices are on channel 11. So if you want to have a mix of Legrand, Xiaomi, Ikea as well as Livolo, it is somehow challenging.

You have nothing to do. you just pair your Livolo on channel 26, and then you can switch Zigate to channel 11, and the Livolo will switch to 11.

crus1313, 2020-02-13:
As I read, Livolo comunicates like 3 times per second its state, so imagine having 10 switches... I believe trafic congestion and interference. I think ch 22 is better.

pipiche38, 2020-06-29:
Should be close

closed 2020-06-30 by fairecasoimeme