fairecasoimeme/ZiGate #149: Devices does not pair with ZiGate

tekaeren, 2019-02-12:
I've finally had time to set up ZiGate Controller - v1.1.
Unfortunately I've problems with pairing Aqara sensors with ZiGate.
I've tested:
- Temp/Press/Humidity sensors
- door sensor

For both I can not get pairing done, as described in https://zigate.fr/getting-started/ -after 5s blinking on sensors there is single blink meaning that pairing has failed, and none information can be seen in ZigBee testing application RawData.
ZiGate is in pairing mode (blinking blue.

I've already tested v30f, v30e and original v30b firmware.

Also with sniffer firmware and Ubiqua Protocol Analyzer no messages can be seem.

Any clues, or ideas how to debug further?

fairecasoimeme, 2019-02-13:
Hi,
Did you erase the eeprom after flashing the ZiGate?
If not. retest and tell me.

If ZiGate blue led blinking, there is no reason but perhaps, the radio module is, perhaps, breaked

tekaeren, 2019-02-14:
Yes, I've been erasing whole EEPROM.
Going both way v30b->v30f and then going back v30f->v30e->v30b

What if radio is broken? Still covered by warranty (2years?)

pipiche38, 2019-02-14:
Don't you have to Start NWK after a full Erase ?

tekaeren, 2019-02-14:
Just to have everything in one place, below the whole procedure for ZiGate USB:
1. Enable flash mode (Zigate key pressed while connectig)
2. In NXP flash programmer Erase EEPROM: Complete EEPROM
3. FLash v30f
4. Reconnect
5. Run ZGWUI test app
6. Connect COM
7. GetVersion: OK;

`Type: 0x8000 (Status)
  Length: 5
  Status: 0x00 (Success)
  SQN: 0x00
  Message: 
Type: 0x8010 (Version)
  Length: 5
  Application: 3
  SDK: 783`

8. Set CMKS: 11
9. Set Type Coordinator
10. Start NWK: Success, NetworkUP
11. Permit Join with mask: FFFC, Time: FE -> Led blinks on device
12. Try to pair, many time for more the minute each. FAIL, oh no!!! Wait it worked.

I think I know what I was doing wrong now... The values You enter in test app is not enough. You need to click LABEL (button in fact) next to them to apply. Dummy me.

Next step connect it to HA.

closed 2019-02-14 by tekaeren