fairecasoimeme/ZiGate #151: ZiGate use LQI not RSSI !

doudz, 2019-02-13:
According to this https://zigate.fr/wiki/protocole-zigate/ each response from ZiGate contains a RSSI (Received Signal Strength Indication) but RSSI is supposed to be a negative value in dBm
source : https://fr.wikipedia.org/wiki/Received_Signal_Strength_Indication

Looking into the code I discover that the value sent is LQI (Link Quality Indicator)

https://github.com/fairecasoimeme/ZiGate/blob/master/Module%20Radio/Firmware/src/ZiGate/Source/ZigbeeNodeControlBridge/app_zcl_event_handler.c#L369-L375

LQI is `psEvent->pZPSevent->uEvent.sApsDataIndEvent.u8LinkQuality` and currently used
RSSI is `psEvent->pZPSevent->uEvent.sApsZgpDataIndEvent.u8Rssi `not used

> Received Signal Strength (RSSI) is a measure of incoherent (raw) RF power in a channel. 
> The RF power can come from any source: other IEEE802.15.4 transmitters, Bluetooth, WiFi, FSK, or background radiation. RSSI is also called Energy Detection (ED) Level and is used for Clear Channel Assessment (CCA). For a single-hop network, RSSI may be used for LQI.
> 
> Link Quality Indication (LQI) is a cumulative value used in multi-hop networks to assess the cost of a link. LQI values can be modified at each step as the massage propagates through the mesh matrix. Messages taking different paths through a mesh may have different LQI values, the path with the best LQI is chosen for subsequent messages. Outgoing LQI values can be calculated from coherent (demodulated) Packet Error Rates (PER), RSSI and incoming LQI values. The exact method depends on the network stack being used.

doudz, 2019-02-13:
So providing LQI instead of RSSI is not really problem but the doc should use the right words

ISO-B, 2019-02-13:
Noticed this same thing in neighbor table yesterday and now that you said it I really understand it.

fairecasoimeme, 2019-02-13:
Right, i'll change doc.

pipiche38, 2019-02-21:
From the NXP documentation , this is somehow contradicting the following statement, So I'm not sure that LQI implementation is not RSSI.

See page 254 of https://www.nxp.com/docs/en/user-guide/JN-UG-3113.pdf
> u8LinkQuality is an LQI value indicating the perceived strength of the radiosignal which carried the received data packet

In regards to the statement that you extract : 
> Link Quality Indication (LQI) is a cumulative value used in multi-hop networks to assess the cost of a link. LQI values can be modified at each step as the massage propagates through the mesh matrix.

As my understanding from the NXP doc, is higher is the LQI better is the Quality, no ?

NXP is also confirming page 70

> LQI value (detected signal strength).

pipiche38, 2019-03-11:
Just had a look to a ZOE remote control for Profalux shutter and as the NXP documentation they talk about RSSI, which seems to do the same as NXP documentation state

pipiche38, 2020-09-30:
This issue has been closed due to no activity since while!

closed 2020-09-30 by fairecasoimeme