fairecasoimeme/ZiGate #152: Eurotronics Thermostic Valve

pipiche38, 2019-02-15:
Just to report the full Zigate compatibility with the Eurotronic Thermostic valve
https://eurotronic.org/produkte/zigbee-heizkoerperthermostat/spirit-zigbee/

Cluster: 0x0201
  * Attributes ( standard ): 
     * 0x0000 -> Local Temperature
     * 0x0010 -> Calibration
     * 0x0012 -> Heating setpoint
     * 0x0015 - >MIN heating setpoint
     * 0x0016 -> MAX heating setpoint
  * Attributes ( specific):
     * 0x4001 -> Valve Position
     * 0x4002 -> Status code
     * 0x4003 -> Current Setpoint ( that is one is either the result of a copy from 0x0012, or is set manually via the Valve head)

There are few others attributes, but didn't find it needed for my purpose

To reset the device (which is needed to pair it to the Zigate), see here https://github.com/sasu-drooz/Domoticz-Zigate/wiki/Thermostats-corner#eurotronic-spirit-zigbee-thermostat-valve--700045-

fairecasoimeme, 2019-02-17:
Thanks for informations. Great job ! ;)

closed 2019-02-18 by pipiche38

jlagrut01, 2019-12-19:
> Just to report the full Zigate compatibility with the Eurotronic Thermostic valve
> https://eurotronic.org/produkte/zigbee-heizkoerperthermostat/spirit-zigbee/
>
> Cluster: 0x0201
>
> * Attributes ( [...]

bonjour,
je n'arrive pas à installer ma vanne thermo Spirit sur le plugin Zigate et le lien ci-dessus ne semble pas fonctionner.
pourriez-vous m'indiquer la démarche à suivre s'il vous plaît ?
merci d'avance,
cordialement,
Jérôme

pipiche38, 2019-12-20:
Quel Plugin Zigate ? pour Domoticz ?

pipiche38, 2019-12-20:
https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Thermostats-corner.md

jlagrut01, 2019-12-22:
merci, j'ai réussi l'installation !

pipiche38, 2019-12-22:
Super!, bon chauffage ;-)