fairecasoimeme/ZiGate #153: Need help building custom firmware

doudz, 2019-02-15:
I'm trying to build a custom firmware in order to support zigpy library.
I manage to build it but trying to using I get in a "Factory New reset" loop.

What I'm trying to do is generate a response 8002 for each message received in APP_ZCL_cbEndpointCallback
https://github.com/fairecasoimeme/ZiGate/blob/master/Module%20Radio/Firmware/src/ZiGate/Source/ZigbeeNodeControlBridge/app_zcl_event_handler.c#L364
To do so, I replace the content of APP_ZCL_cbEndpointCallback by a call to the function 
 Znc_vSendDataIndicationToHost
https://github.com/fairecasoimeme/ZiGate/blob/master/Module%20Radio/Firmware/src/ZiGate/Source/ZigbeeNodeControlBridge/app_general_events_handler.c#L1347

Since I'm not a C/C++ guru, any help will be appreciate :smile:

ISO-B, 2019-02-16:
Did you managed to get correct parameters for Znc_vSendDataIndicationToHost? There is no reason that I  could quickly see that would result Factory New Reset loop, except if your parameters for that function are bad and ZiGate crashes and resets.

doudz, 2019-02-16:
Thanks for advice, I'll check my code

doudz, 2019-02-20:
This is what I made :
https://github.com/doudz/ZiGate-1/blob/raw_mode/Module%20Radio/Firmware/src/ZiGate/Source/ZigbeeNodeControlBridge/app_zcl_event_handler.c#L378-L383

ISO-B, 2019-02-20:
Ok. I will test it tomorrow with uart debug so I can see what is happening.

ISO-B, 2019-02-21:
https://github.com/doudz/ZiGate-1/blob/6caeb9df0a1fda35f61ff41e334744e4ca5570a1/Module%20Radio/Firmware/src/ZiGate/Source/ZigbeeNodeControlBridge/app_zcl_event_handler.c#L381

Change that to `Znc_vSendDataIndicationToHost(&psEvent->pZPSevent, au8LinkTxBuffer);`

Also add 
`PUBLIC void Znc_vSendDataIndicationToHost ( ZPS_tsAfEvent* psStackEvent, uint8* pau8LinkTxBuffer );`
 before https://github.com/fairecasoimeme/ZiGate/blob/76c328bb44a6fdd1774c86807b101c9c15edfebc/Module%20Radio/Firmware/src/ZiGate/Source/Common/app_common.h#L209

doudz, 2019-02-21:
Thank you so much !
I don't know about the **&**

But it's already in app_common.h, isn't it ?
https://github.com/fairecasoimeme/ZiGate/blob/76c328bb44a6fdd1774c86807b101c9c15edfebc/Module%20Radio/Firmware/src/ZiGate/Source/Common/app_common.h#L189

doudz, 2019-02-23:
Now it works, but not as expected. I cet response 0x8002 but the content has no sense

doudz, 2019-03-06:
as I said, I get response 0x8002 now but the content has no sense
it's supposed to be a attribute report of a xiaomi square button, so onoff cluster 0x0006 attribute 0x0000 
I get the following :

`018002120210021e021a0214021402104558a4021046021002100214021002107d03018002120210021e021a0214021402104558a4021046021002100214021002107d03018002120210021e021a0214021402104558a4021046021002100214021002107d03`

so 3 times the same `018002120210021e021a0214021402104558a4021046021002100214021002107d03`

If I decode it `018002000e0a0404004558a4004600000400007d03`
msg_type : 8002
length : 000e (14 so looks good)
checksum 0a 
status 04 ?
Profile 0400 ?
cluster 4558 ? 
source endpoint a4 ?
destination endpoint 00 ? 
source addr mode 46 ?
source addr 0000 ?
destination addr mode 04 ?
destination addr 0000 ?
lqi 7d

Znc_vSendDataIndicationToHost doesn't work as expected but I don't know why
Any help will be appreciate

ISO-B, 2019-03-06:
Sorry this has been partially my fault. You should have used `Znc_vSendDataIndicationToHost(psEvent->pZPSevent, au8LinkTxBuffer);` without &.

Using it without it will make Znc_vSendDataIndicationToHost to crash to line `uint8*    dataPtr   =  ( uint8* ) PDUM_pvAPduInstanceGetPayload ( psStackEvent->uEvent.sApsDataIndEvent.hAPduInst );`

Will I try to fix SendDataIndicationToHost tomorrow and get back to you. It might be that not all messages have psStackEvent->uEvent.sApsDataIndEvent.hAPduInst and that's way it crashes. You could try commenting lines that use dataPtr and u8Size out and see if its working then.

doudz, 2019-03-06:
Thanks for your reply, I have put the **&** , it crash without.
I'll try with commenting the payload lines and see

ISO-B, 2019-03-07:
Tested this and noticed that not all events have hAPduInst which is why it crashes. Good solution might be to filter messages based psEvent->eEventType's that you want to get feedback and forward those to  Znc_vSendDataIndicationToHost. There are multiple event types that are not anything that user needs to know like . More info about event types can be found from JN-UG-3103 Chapter 35.3 ZCL Event Enumerations.

```
ZPS_tsAfEvent* psStackEvent = psEvent->pZPSevent;
if (psEvent->eEventType != E_ZCL_CBET_CLUSTER_UPDATE && psEvent->eEventType != E_ZCL_CBET_UNHANDLED_EVENT )
{
    Znc_vSendDataIndicationToHost(psStackEvent, au8LinkTxBuffer);
}
```
That will get you at least started. Won't crash at start and managed to pair device without crash also.

doudz, 2019-03-07:
Really thank you to took time to test.
Maybe Znc_vSendDataIndicationToHost is not what I need, I have to go deeper in zigpy

ISO-B, 2019-03-07:
Do you really need all events? I mean including all internal timers, locks and option changes.

doudz, 2019-03-07:
No I don't think, but the first one is attribute change, I'll test with your code

pipiche38, 2020-09-30:
This issue has been closed due to no activity since while!

closed 2020-09-30 by fairecasoimeme