fairecasoimeme/ZiGate #155: Firmaware 3.0f - Ikea Tradfri

battousai90, 2019-02-16:
Hello,

Since the frimware upgrade to 3.0f, the temperature control of my Ikea Tradfri don't work.

fairecasoimeme, 2019-02-17:
Hi,
we did not make any changes on that part. Are you sure this is the only modification ? no plugin update ? no other manipulations ?

battousai90, 2019-02-17:
Hello,

Only flash for firmware 3.0f, so of course eeprom erase and include all device.
Before i was on Jeedom with Plugin 1.2.2 and firmware 3.0f, and no problem, all was perfect. Jeedom send me notification for said that Zigate firmware is outdated, so i checked for upgrade it.
After the flash to 3.0f, i still on Jeedom with plug-in 1.2.2, saw all device but nothing work, after an erase PDM all device was disappaired. Include all again,but Tradfri not have Temperature control, i have the slider, but nothing happen when move it.

battousai90, 2019-02-17:
After some test, the problem is the plugin Zigate 1.2.2, because with the plug-in Abeille, temperature controle on IKEA tradfri works perfectly.

closed 2019-02-17 by fairecasoimeme

doudz, 2019-02-18:
There was a bug in version 1.2.* of the Jeedom plugin. It has just been fixed and it should be ok in the next 1.2.3 release

battousai90, 2019-02-19:
Ok thank you very much Doudz. Waiting on 1.2.3.