fairecasoimeme/ZiGate #157: Address Mode and Command

pipiche38, 2019-02-18:
The NXB Zigebee stack allow to specify Device address based on the address mode:
- Network address ( short address) - uint16
- IEEE  uint64

If I look to Bind/Unbind command ( 0x0030/0x0031 ) it is pretty clear that we can use either IEEE or NetworkAddress

When looking at the Group, Identify, Action Move, Action ON/FF ....  There is an Address Mode to be specify , BUT the target address seems to be only an uint16

Could you kindly clarify, when address mode is available we can specify which address mode we want to use and so the address will be based on the address mode (as documented for Bind/UBIND). 
In case there is not option to use anything else than a short address, then the address mode is meaningless and from the documentation standpoint it must be written that the ONLY possible value is 0x02

ISO-B, 2019-03-08:
It would be incorrect to say that ONLY possible value is 0x02 even if we ignore couple of special cases like binds. Generally you can use address modes 0x01(Group) and 0x02(Short address), which are both 16 bit addresses. Can you tell me case when you would like to use IEEE address instead of short address? ZiGate should keep track of short and long address pairs anyway.

ISO-B, 2019-03-19:
All cluster commands should support 64-bit address. At the moment parameter placements are hard coded in APP_vProcessIncomingSerialCommands. Because of that it's not just possible to quickly enable 0x03 for all commands. We need to add variable that will determinate start position of other parameters based on address mode.

pipiche38, 2020-09-30:
to be closed . Implemented in 31d

closed 2020-09-30 by pipiche38