fairecasoimeme/ZiGate #159: ZigateWifi Crash

yannatparis, 2019-02-22:
Hello, 

After configure it, the Zigate wifi connects to the network. But after a few seconds. it stop being connect.
I cannot ping it again, it doesn't even reappear as "access point".
it was the same each time i rebooted it. (connect to my network then disappear)

the strange things is , if i remove the Radio Controller. The wifi module works fine alone.
i can ping it as long as i want to.

I tried with different USB power plug.

Zigate wifi v1.3
Firmware of radio controller 3.0E

Best, 
Yann

fairecasoimeme, 2019-06-03:
Hi,
Sorry for delay but you have to do 2 things:

First, update the ZiGate to v3.0f
Second, I recommand to you to update WiFi module with this procedure :
[https://zigate.fr/documentation/installation-du-firmware-v2-xx/](https://zigate.fr/documentation/installation-du-firmware-v2-xx/)
and 
[https://zigate.fr/documentation/parametrer-le-module-wifi-2/](https://zigate.fr/documentation/parametrer-le-module-wifi-2/)

Fred

yannatparis, 2019-07-27:
Hi Fred,

it works, thx

Best,
Yannick

closed 2019-08-09 by fairecasoimeme