fairecasoimeme/ZiGate #160: Is the zigbee remote  INNR RC110 supported?

shupa, 2019-02-23:
Looking for a zigbee remote where I can have at least 5 physical buttons (as my main goal is to control independently each lights), I found this one but not sure I will be able to pair it with my zigate.
Any idea/clue?

Thanks!

ISO-B, 2019-02-23:
That looks quite interesting remote. I could try to find one for cheap so I could experiment with it. Based on very quick research it seems that it would need some work to get is working. It might be possible to get it working with gateway without directly pairing it with any lamps. 

More general discussion about this remote can be hound https://github.com/dresden-elektronik/deconz-rest-plugin/issues/635

fairecasoimeme, 2019-02-23:
I ordered one too

shupa, 2019-02-24:
Looks like it will be supported soon ;)
Thanks a lot guys.

pipiche38, 2019-03-09:
At least it is seen by the Zigate. You can paired it.

Here after are the Endpoints and related clusters
'Ep': {'01': {'0000': {}, '0004': {}, '0003': {}, '0006': {}, '0008': {}, '0005': {}, '0019': {}}, 
'02': {'1000': {}}, 
'03': {'0004': {}, '0005': {}, '0003': {}, '0006': {}, '0008': {}, }, 
'04': {'0004': {}, '0005': {}, '0003': {}, '0006': {}, '0008': {}}, 
'05': {'0004': {}, '0005': {}, '0003': {}, '0006': {}, '0008': {}}, 
'06': {'0004': {}, '0005': {}, '0003': {}, '0006': {}, '0008': {}}, 
'07': {'0004': {}, '0005': {}, '0003': {}, '0006': {}, '0008': {}}, 
'08': {'0004': {}, '0005': {}, '0003': {}, '0006': {}, '0008': {}}

pipiche38, 2019-03-11:
Funny remote, while pairing it on my prod network. seems that all devices with cluster 0x0006 and 0x0008 have been retrieve by the remote, without creating a Group.

I was able on the 'scene' position able to On/off (bulbs, switches and shutters) in once. Same on level control.

I did a scan of group membership after the pairing , nothing discovered. No new groups and in reality it was a fresh network, so no Group at all !

shupa, 2019-03-22:
Seems a bit too technical for me :)
Does it mean it will be supported and used as I would like? (individually control lights and/or plugs)?

Thanks guys!

pipiche38, 2019-05-03:
#194

pipiche38, 2019-05-03:
It looks like the Remote is sending broadcast command. So all devices with cluster 
0x0006 respond to the Power On/Off of the remote
0x0008 respond to the Level Control - and +
0x0008 respond to the Scene 1,2,3,4,5,6

pipiche38, 2019-05-03:
Some information

```   
 "00a3": {
        "App Version": "34",
        "Battery": {},
        "Ep": {
            "01": {
                "0000": {},
                "0003": {},
                "0004": {},
                "0005": {},
                "0006": {},
                "0008": {},
                "0019": {}
            },
            "02": {
                "1000": {},

            "03": {
                "0003": {},
                "0004": {},
                "0005": {},
                "0006": {},
                "0008": {}
            },
            "04": {
                "0003": {},
                "0004": {},
                "0005": {},
                "0006": {},
                "0008": {}
            },
            "05": {
                "0003": {},
                "0004": {},
                "0005": {},
                "0006": {},
                "0008": {}
            },
            "06": {
                "0003": {},
                "0004": {},
                "0005": {},
                "0006": {},
                "0008": {}
            },
            "07": {
                "0003": {},
                "0004": {},
                "0005": {},
                "0006": {},
                "0008": {}
                }
            },
            "08": {
                "0003": {},
                "0004": {},
                "0005": {},
                "0006": {},
                "0008": {}
            }
        },
        "HW Version": "2",
        "IEEE": "00158d00018bee82",
        "MacCapa": "80",
        "Model": "RC 110",
        "NbEp": "8",
        "ProfileID": "c05e",
        "Stack Version": "14",
        "Version": "3",
        "ZDeviceID": "0820"
    },

```

pipiche38, 2019-07-09:
Thanks @fairecasoimeme , it works ! Now we just have the manage the Remote from the User space.

pipiche38, 2019-07-14:
Can be close. Fully addressed in 3.1a

closed 2019-07-15 by ISO-B

shupa, 2019-07-22:
Anything special to link the remote with the zigate?
I tried several times by (as mentioned) keeping simultaneously the  « – » and « program » for at least 10s with of course zigate in discovery mode and nothing happen (latest firmware + latest plugin zigate jeedom).
Any tips?

pipiche38, 2019-07-22:
I suggest that you refer to the manual on how to reset the remote and then do the pairing with Zigate
https://www.innrlighting.com/clients/innr/themes/uploads/1484932356_Manual%20for%20Remote%20Control%20-%20RC%20110.pdf

Indeed to get it fully handled by the firmware your must have 3.1a

> You    can    reset    the    remote    by    simultaneously    pressing    and    holding    the    "Program"    button    and    button    for    5seconds.    When    the    remote    has    been    reset,    the    green    LED    will    blink    5    times.

shupa, 2019-07-23:
Thanks! Working ;)