fairecasoimeme/ZiGate #162: some commands are unavailable because require FULL_FUNC_DEVICE compilation

doudz, 2019-02-24:
https://github.com/fairecasoimeme/ZiGate/blob/76c328bb44a6fdd1774c86807b101c9c15edfebc/Module%20Radio/Firmware/src/ZiGate/Source/ZigbeeNodeControlBridge/app_Znc_cmds.c#L698-L736

affected commands :
factory reset 0x0013
touchlink 0x00D0
touchlink factory reset 0x00D2

Maybe there's other

ISO-B, 2019-02-24:
All of those are touchlink related.

doudz, 2019-02-24:
Except the Factory reset 0x0013, isn't it

ISO-B, 2019-03-07:
It is touchlink related because it won't do anything as long as device is coordinator. On the other hand the function that it calls resets device data. I would use 0x0012 which erases PDM and reboots ZiGate. I can't think any case where it would be necessarily to reset ZiGate without rebooting it.

doudz, 2019-03-08:
Ok so 0x0013 is unneeded.
And what about touchlink functions ? 
What's the purpose of Touchlink factory reset ? it it for resetting light (for example) like the Hue Remote ?

pipiche38, 2019-03-08:
+1 to follow the Touchlink factory reset

Syxxxx, 2019-03-18:
Agree for adding capability of Touchlink factory reset

pdecat, 2019-12-30:
Was this supposed to be included with firmware v3.1a?

I still get unhandled command errors when executing 0x00D0 / Initiate Touchlink:

```
2019-12-30 17:29:02 DEBUG (SyncWorker_1) [zigate] REQUEST : 0x00d0 
2019-12-30 17:29:02 DEBUG (SyncWorker_1) [zigate] Msg to send b'00d00000d0'
2019-12-30 17:29:02 DEBUG (SyncWorker_1) [zigate] Encoded Msg to send b'010210d002100210d003'
2019-12-30 17:29:02 DEBUG (SyncWorker_1) [zigate] Waiting for status message for command 0x00d0
2019-12-30 17:29:02 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15W\x02\x12\x02\x10\x02\x10\xd0\x02\x10\x03'
2019-12-30 17:29:02 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-12-30 17:29:02 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8000: b'020000d0'
2019-12-30 17:29:02 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8000 - Status response : status:2, sequence:0, packet_type:208, error:b'', lqi:0
2019-12-30 17:29:02 ERROR (ZiGate-Decode data) [zigate] Command 0x00d0 failed Unhandled command : b''
2019-12-30 17:29:02 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-12-30 17:29:02 DEBUG (SyncWorker_1) [zigate] STATUS code to command 0x00d0:RESPONSE 0x8000 - Status response : status:2, sequence:0, packet_type:208, error:b'', lqi:0
```

Same with 0x00D2 / Touch link factory reset target.

pipiche38, 2019-12-31:
I don’t think any touch link features are enabled on zigate

pipiche38, 2020-09-30:
This issue has been closed due to no activity since while!

closed 2020-09-30 by fairecasoimeme