fairecasoimeme/ZiGate #163: Zigate doc and 0x8000 message

pipiche38, 2019-02-25:
In Zigate.fr for the description of Réponse 0x8000, I see that there is an Optional additional error information (as a string).

Could you kindly clarify what should we expect here ?

ISO-B, 2019-03-18:
I took quick look on source and couldn't find any instance where 0x8000 would return anything else than status, sequence number and packet type. Unless there is internal commands that we can't see. 

@fairecasoimeme do you know for sure that there is case where it will output optional error information or is this just for future proofing?

pipiche38, 2019-03-18:
Obj -> ZiGate | 0x8000 | Status | <status:uint8_t> | All status messages will have a sequence number sent back. Default of 0 for messages which are not transmitted over the air.
-- | -- | -- | -- | --
  |   |   | <sequence number: uint8_t> |  
  |   |   | <Packet Type: uint16_t> |  
  |   |   | <Optional additional error information: string>

fairecasoimeme, 2019-03-19:
I copied informations from JN-AN-1216-ZigBee-3-0-IoT-ControlBridge-UserGuide.pdf
But, It's true, I never seen Option all error information yet.

ISO-B, 2019-03-20:
Maybe it is there so users could implement their own additional error information. Remove it from documentation for now and add back if it becomes useful?

fairecasoimeme, 2019-03-27:
Updated

closed 2019-03-27 by fairecasoimeme