fairecasoimeme/ZiGate #164: No 0x8000 response for 0x0011 message ?

pipiche38, 2019-02-27:
From ZGUI when you do a (Soft Reset) 0x0011 you don't get Status message ! From the documentation ( Zigate.fr Command) this is expected.

When doing it outside of ZGUI , there is indeed a 0x8000 message, which seems to be malformed. Bad lenght/ Bad CRC.

ZGUI

13:44:35.896 -> 01 02 10 11 02 10 02 10 11 03
13:44:36.065 <- 01 80 24 00 0D E0 01 00 00 00 15 8D 00 01 ED ED DE 0F 00 03
13:44:36.147 <- 01 80 06 00 02 85 01 00 03

Looks like ZGUI didn't display 0x8000 !! Does it get also a CRC/FRAME error ?

pipiche38, 2019-02-27:
https://github.com/sasu-drooz/Domoticz-Zigate/issues/369

pipiche38, 2019-03-06:
Problem exist also on PI-Zigate

ISO-B, 2019-03-18:
This probably happens because reset happens before ZiGate has time to send whole response. I will investigate and try to fix this.

pipiche38, 2019-03-18:
Good news

closed 2019-03-21 by pipiche38

pipiche38, 2019-06-22:
Confirmed

commit 9348aa5f (2019-03-27): Added delay for reset after reset command

ZiGate reseted before it had time to sent 0x8000 response after reset
command. Fixes #164