fairecasoimeme/ZiGate #165: Tradfri OTA update problem

Jsl-1, 2019-02-28:
Hi, 

I tried to update an ikea tradfri outlet using ota file.
Step : 
  - Download ota image from ikea http://fw.test.ota.homesmart.ikea.net/Tradfri_OTA_release_signed_2019_01_30_154434/bin/10005777-3.1-TRADFRI-control-outlet-2.0.019.ota.ota.signed
  - Convert ota file with deCONZ V2_05_20 (Windows)
  - Move file to my home assistant installation using smb
  - use python library to call ota_load_image('10005777-3.1-TRADFRI-control-outlet-2.0.019.ota.ota.signed')

I get this: Something wrong with ota file header.
I use python library 2.8.0 and zigate USB TTL with firmware 3.0f

Here is the debug log for command 0x0500: 

DEBUG:zigate:REQUEST : 0x0500 b'\x02\x00\x00\x0b\xee\xf1\x1e\x01\x00\x008\x00\x00\x11|\x11\x01 \x01\x96#\x00\x02EBL tradfri_control_outlet      \x00\x03\x19\xbe\x00\x00\x00\x00\x00\x03\x19\x80\x00\x02\x8c\xe3\x02'
DEBUG:zigate:Msg to send b'05000048a70200000beef11e010000380000117c110120019623000245424c20747261646672695f636f6e74726f6c5f6f75746c6574202020202020000319be000000000003198000028ce302'
DEBUG:zigate:Encoded Msg to send b'0102150210021048a7021202100210021beef11e0211021002103802100210117c11021120021196230210021245424c20747261646672695f636f6e74726f6c5f6f75746c65742020202020200210021319be0210021002100210021002131980021002128ce3021203'
DEBUG:zigate:Waiting for status message for command 0x0500
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15f\x02\x10\xd3\x02\x100\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'00d30030'
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:211, packet_type:48, error:b'', lqi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0030:RESPONSE 0x8000 - Status response : status:0, sequence:211, packet_type:48, error:b'', lqi:0
DEBUG:zigate:Waiting for message 0x8030
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\x80\x02\x10\x02\x10\x02\x15\x02\x10\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'00000500'
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:1280, error:b'', lqi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0500:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:1280, error:b'', lqi:0
WARNING:zigate:Something wrong with ota file header.
WARNING:zigate:Cannot read ota header. No ota file loaded.

I tried with other files from http://fw.ota.homesmart.ikea.net with same result
Did I miss something ?

Thank you for your help.

_Originally posted by @Jsl-1 in https://github.com/fairecasoimeme/ZiGate/issues/86#issuecomment-466911946_

ISO-B, 2019-03-01:
Problem is with python library. Will make fix for it.

Jsl-1, 2019-03-01:
Ok, thank you very much ;)

closed 2019-03-01 by Jsl-1

doudz, 2019-03-01:
Fixed in 0.28.4 Thanks @ISO-B