fairecasoimeme/ZiGate #167: Attibute map16 problem

Mukatai, 2019-03-04:
Hi,

When i request a map16 attribute (Type 0x19 : 16-bit bitmap - BITMAP16) i received an attribute with size 0 :

`Type: 0x8102 (Attribute Report)
  SQN: 0x14
  Src Addr: 0x3788
  Src Ep: 0x01
  Cluster ID: 0x0300 (Lighting: Color Control)
  Attribute ID: 0x400A
  Attribute Size: 0x0000
  Attribute Type: 0x19 (Unknown)
  Status: 0x00
`

`01 81 02 00 0D FE 14 37 88 01 03 00 40 0A 00 19 00 00 8A 03`

If i follow the structure response i have not the good bitmap and 1 for the LQI. (I think is the start of next command). The data byte list does not seem to be in the response.

@+

pipiche38, 2019-04-17:
Si je decode la Frame qui est donné, je trouve bien un 16bit bitmap. Non ?

Start Frame: 01 
     Command: 81 02 
     Lenght: 000D
     CRC: FE 
Payload:
     SQN: 14 
     ShortAddr: 37 88 
     EP: 01 
     ClusterId: 03 00 
     Attribute: 40 0A 
     Status: 00
     Data Type:  19 
     Attribute Data: 00 00 
RSSI: 8A 
End Frame: 03

Mukatai, 2019-04-18:
Je n'avais pas détaillé le problème. Il y a plusieurs problèmes : 
- La taille de l'attribut renvoyée est 0.
- Aucun byte n'est renvoyé alors qu'il devrait en y avoir 2 pour ce type.

Si je décode selon la documentation j'obtient cela :
Sequence number (uint8_t) : **14**
Src address (uint16_t) : **37 88**
Endpoint: (uint8_t) : **01**
Cluster id (uint16_t) : **03 00**
Attribute Enum (uint16_t) : **40 0A**
Attribute status (uint8_t) : **00**
Attribute data type (uint8_t) : **19**
Size Of the attributes in bytes (uint16_t) : **00 00** 
Data byte list (stream of uint8_t) : **8A 03**

On voit que le lqi (8A) et le byte de fin (03) se retrouve dans les byte du map 16. Jusqu'au type tout est bon mais c'est apres que ca ne correspond plus. J'obtient ce résultat avec un lampe Ikea Couleur sur le cluser 0x300 et l'attribute 0x400A (ColorCapabilities map16)

pipiche38, 2019-04-18:
Exact, j'avais oublié le 'Size' dans mon décodage.

A voir avec @fairecasoimeme , mais si la taille est Nulle, c'est certainement normalement de ne pas avoir de data. Donc pour moi la trame est correcte.

La question est plutôt, est-ce coté Ikea ou Zigate qu'il y a un problème. Un sniffeur serait le bien venu pour voir exactement ce qui transite.

Mukatai, 2019-04-18:
Oui c'est clair que le deux problème sont liés. Après oui il faut valider ou non que la lampe renvoi quelque chose de correct sur cette commande. Faudrait que je test avec une Philips ou autre marque pour voir si j'ai le meme résultat. De mémoire j'obtient le meme résultat avec une Ikea où on peu modifier la température. Il y a aussi le cluster 0x0500 où il y a un map16 sur l'attribut 0x0002, a voir si on l'obtient bien quand on le demande…

Je n'ai rien pour l'instant pour sniffer :(

fairecasoimeme, 2019-04-25:
Salut,
Peux-tu me donner la référence de ton/tes appareils pour que je puisse faire des tests ?
De premier abord, si "attribute size" est égal à zéro tout est normal ...

Merci

Mukatai, 2019-04-25:
TRADFRI bulb E27 WS opal 980lm
TRADFRI bulb E27 CWS opal 600lm

Model renvoyé par le cluster basic...

fairecasoimeme, 2019-04-25:
Ok merci, je vais en commander.

Mukatai, 2019-04-29:
👍 
Merci, je testerai quand j'aurai un peu de temps…

pipiche38, 2019-07-06:
A priori ca fonctionne sur le Cluster 0x0500 avec attribute 0x002
ReadCluster0500 - Security & Safety IAZ Zone - Device: 0771 
     MsgAttrID: 0002 
     MsgAttType: 19 
     MsgAttSize: 0002 
     MsgClusterData: 0010

Infos cohérente

KiwiHC16, 2019-07-09:
With V3.1a, I can confirm the info is Ok.

closed 2019-07-17 by fairecasoimeme