fairecasoimeme/ZiGate #169: Get Serval Attribute Problem

Mukatai, 2019-03-06:
Hi,

I have this problem only with Basic cluster on Legrand Devices.

When i request several attribute on the basic cluster i receive an invalid attribute on 0x0005 attribute. But if i request only this attribute or less attribute it's ok.

**Exemple KO :**

Request 0x0000 0x0001 0x0002 0x0003 0x0004 0x0005 :

`01 02 11 02 10 02 10 18 F2 02 12 DE 30 02 11 02 11 02 10 02 10 02 10 02 10 02 10 02 10 02 16 02 10 02 10 02 10 02 11 02 10 02 12 02 10 02 13 02 10 02 14 02 10 02 15 03`

Receive :

`01 01 80 00 00 05 8F 00 0B 01 00 00 03`
`01 01 81 02 00 0E D5 0B DE 30 01 00 00 00 00 00 20 00 01 02 9F 03`
`01 01 81 02 00 0E D6 0B DE 30 01 00 00 00 01 00 20 00 01 00 9F 03`
`01 01 81 02 00 0E 95 0B DE 30 01 00 00 00 02 00 20 00 01 40 9F 03`
`01 01 81 02 00 0E D5 0B DE 30 01 00 00 00 03 00 20 00 01 01 9F 03`
`01 01 81 02 00 2C FA 0B DE 30 01 00 00 00 04 00 42 00 1F 20 4C 65 67 72 61 6E 64 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 9F 03`
`01 01 81 02 00 0D 86 0B DE 30 01 00 00 00 05 89 FF 00 00 9F 03`

> Attribute Message (0x8102) 
 Length : 14 
 Source : 0xDE30 
 Endpoint : 0x01 
 Cluster : 0x0000 (General: Basic) 
 Data Size : 1 
 Attribute Enum : 0x0000 
 Attribute Status : 0x00 
 Attribute Type : 0x20 (Unsigned 8-bit int) 
 Attribute Size : 1 
 Attribute Data : 2 
 LQI : 159

> Attribute Message (0x8102) 
 Length : 14 
 Source : 0xDE30 
 Endpoint : 0x01 
 Cluster : 0x0000 (General: Basic) 
 Data Size : 1 
 Attribute Enum : 0x0001 
 Attribute Status : 0x00 
 Attribute Type : 0x20 (Unsigned 8-bit int) 
 Attribute Size : 1 
 Attribute Data : 0 
 LQI : 159

> Attribute Message (0x8102) 
 Length : 14 
 Source : 0xDE30 
 Endpoint : 0x01 
 Cluster : 0x0000 (General: Basic) 
 Data Size : 1 
 Attribute Enum : 0x0002 
 Attribute Status : 0x00 
 Attribute Type : 0x20 (Unsigned 8-bit int) 
 Attribute Size : 1 
 Attribute Data : 64 
 LQI : 159

> Attribute Message (0x8102) 
 Length : 14 
 Source : 0xDE30 
 Endpoint : 0x01 
 Cluster : 0x0000 (General: Basic) 
 Data Size : 1 
 Attribute Enum : 0x0003 
 Attribute Status : 0x00 
 Attribute Type : 0x20 (Unsigned 8-bit int) 
 Attribute Size : 1 
 Attribute Data : 1 
 LQI : 159

> Attribute Message (0x8102) 
 Length : 44 
 Source : 0xDE30 
 Endpoint : 0x01 
 Cluster : 0x0000 (General: Basic) 
 Data Size : 31 
 Attribute Enum : 0x0004 
 Attribute Status : 0x00 
 Attribute Type : 0x42 (Character String) 
 Attribute Size : 31 
 Attribute Data : Legrand 
 LQI : 159

> Attribute Message (0x8102) 
 Length : 13 
 Source : 0xDE30 
 Endpoint : 0x01 
 Cluster : 0x0000 (General: Basic) 
 Data Size : 0 
 Attribute Enum : 0x0005 
 Attribute Status : 0x89 
 Invalid Attribute

**Exemple OK :**

Request  0x0003 0x0004 0x0005 :

`01 02 11 02 10 02 10 12 FE 02 12 DE 30 02 11 02 11 02 10 02 10 02 10 02 10 02 10 02 10 02 13 02 10 02 13 02 10 02 14 02 10 02 15 03`

Receive :

`01 01 81 02 00 0E E6 0F DE 30 01 00 00 00 03 00 20 00 01 01 A8 03`
`01 01 81 02 00 2C C9 0F DE 30 01 00 00 00 04 00 42 00 1F 20 4C 65 67 72 61 6E 64 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 A8 03`
`01 01 81 02 00 2C 89 0F DE 30 01 00 00 00 05 00 42 00 1F 20 52 65 6D 6F 74 65 20 73 77 69 74 63 68 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 A8 03`

> Attribute Message (0x8102) 
 Length : 14 
 Source : 0xDE30 
 Endpoint : 0x01 
 Cluster : 0x0000 (General: Basic) 
 Data Size : 1 
 Attribute Enum : 0x0003 
 Attribute Status : 0x00 
 Attribute Type : 0x20 (Unsigned 8-bit int) 
 Attribute Size : 1 
 Attribute Data : 1 
 LQI : 168

> Attribute Message (0x8102) 
 Length : 44 
 Source : 0xDE30 
 Endpoint : 0x01 
 Cluster : 0x0000 (General: Basic) 
 Data Size : 31 
 Attribute Enum : 0x0004 
 Attribute Status : 0x00 
 Attribute Type : 0x42 (Character String) 
 Attribute Size : 31 
 Attribute Data : Legrand 
 LQI : 168

> Attribute Message (0x8102) 
 Length : 44 
 Source : 0xDE30 
 Endpoint : 0x01 
 Cluster : 0x0000 (General: Basic) 
 Data Size : 31 
 Attribute Enum : 0x0005 
 Attribute Status : 0x00 
 Attribute Type : 0x42 (Character String) 
 Attribute Size : 31 
 Attribute Data : Remote switch 
 LQI : 168

I have the same behavior with the Legrand Micro Module. 
That is a Legrend bug or a Zigate bug ?

@+

fairecasoimeme, 2019-03-09:
the 0x89 status is : A child descriptor was not avalaible following a discovery request to a parent.
I have to do some tests to confirm if this is ZiGate or Legrand devices.

pipiche38, 2019-10-09:
What I have found is :
- Attribute 0x0005 is corretly provided by Legrand devices.

The issue is that it looks like Legrand doesn't accept too many attribute request at the same time! So when doing your Read Attribute Req ( 0x0100 ) you should limit the number of attribute

pipiche38, 2020-09-30:
This issue has been closed due to no activity since while!

closed 2020-09-30 by fairecasoimeme