fairecasoimeme/ZiGate #171: Color problem with Paul Neuhaus Q-LED

doudz, 2019-03-08:
A Jeedom user reports a problem using Paul Neuhaus Q-LED, when trying to change the color, the bulb always replies with status 0x80 (128) Malformed command. The command appears to contain the wrong fields or invalid field.
Any help will be appreciate :smile: 

RESPONSE 0x8101 - Default device response : sequence:2, endpoint:2, cluster:768, cmd:7, status:128, lqi:90

https://github.com/Jeedom-Zigate/jeedom-plugin-zigate/issues/152

fairecasoimeme, 2019-03-09:
Unfortunately, my Q-LED does not handle colors, although in clusters there is color control.
hard to test ...

pipiche38, 2020-09-30:
This issue has been closed due to no activity since while!

closed 2020-09-30 by fairecasoimeme