fairecasoimeme/ZiGate #172: log errors with new firmware Zigate 3.0f

echaplet, 2019-03-08:
Hello,
I updated my module USB-TTL with Flash programmer under windows 10 including the last step of Erase Complete EEPROM. I use the file ZiGate_v30f.bin.

Back into my jeedom box, the new version was good (3.0f) but i cannot include new Aqara sensors and all my zigbee sensors (aqara) our out of order.
Can you help me please ?

See debug log below 👍 
[2019-03-08 11:05:22][DEBUG] : callZiGate permit_join
[2019-03-08 11:05:22][DEBUG][root](Thread-4) : Message received in socket
[2019-03-08 11:05:22][DEBUG][root](Thread-4) : {'args': '', 'action': 'permit_join'}
[2019-03-08 11:05:22][DEBUG][zigate](Thread-4) : REQUEST : 0x0049 FFFC1E00
[2019-03-08 11:05:22][DEBUG][zigate](Thread-4) : Msg to send b'0049000450fffc1e00'
[2019-03-08 11:05:22][DEBUG][zigate](Thread-4) : Encoded Msg to send b'010210490210021450fffc1e021003'
[2019-03-08 11:05:22][DEBUG][zigate](Thread-4) : Waiting for status message for command 0x0049
[2019-03-08 11:05:22][DEBUG][zigate](ZiGate-Listen) : Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\x19\x02\x10\xd5\x02\x10I\x02\x10\x03'
[2019-03-08 11:05:22][DEBUG][zigate](ZiGate-Event Loop) : Dispatch ZIGATE_PACKET_RECEIVED
[2019-03-08 11:05:22][DEBUG][zigate](ZiGate-Decode data) : Received response 0x8000: b'00d50049'
[2019-03-08 11:05:22][DEBUG][zigate](ZiGate-Decode data) : RESPONSE 0x8000 - Status response : status:0, sequence:213, packet_type:73, error:b'', lqi:0
[2019-03-08 11:05:22][DEBUG][zigate](ZiGate-Decode data) : Dispatch ZIGATE_RESPONSE_RECEIVED
[2019-03-08 11:05:22][DEBUG][zigate](Thread-4) : STATUS code to command 0x0049:RESPONSE 0x8000 - Status response : status:0, sequence:213, packet_type:73, error:b'', lqi:0
[2019-03-08 11:05:22][DEBUG][root](Thread-4) : {'result': OrderedDict([('status', 0), ('sequence', 213), ('packet_type', 73), ('error', b''), ('lqi', 0)]), 'success': True}
[2019-03-08 11:05:22][DEBUG] : result callZiGate Array (     [result] => Array         (             [status] => 0             [sequence] => 213             [packet_type] => 73             [error] =>              [lqi] => 0         )      [success] => 1 )
[2019-03-08 11:05:27][DEBUG] : callZiGate get_version_text
[2019-03-08 11:05:27][DEBUG][root](Thread-4) : Message received in socket
[2019-03-08 11:05:27][DEBUG][root](Thread-4) : {'args': '', 'action': 'get_version_text'}
[2019-03-08 11:05:27][DEBUG][root](Thread-4) : {'result': '3.0f', 'success': True}
[2019-03-08 11:05:27][DEBUG] : result callZiGate Array (     [result] => 3.0f     [success] => 1 )
[2019-03-08 11:05:27][DEBUG] : callZiGate get_libversion
[2019-03-08 11:05:27][DEBUG][root](Thread-4) : Message received in socket
[2019-03-08 11:05:27][DEBUG][root](Thread-4) : {'args': '', 'action': 'get_libversion'}
[2019-03-08 11:05:28][DEBUG][root](Thread-4) : {'result': '0.28.4', 'success': True}
[2019-03-08 11:05:28][DEBUG] : result callZiGate Array (     [result] => 0.28.4     [success] => 1 )

echaplet, 2019-03-08:
sorry. i forgot to hard reset my sensors...

closed 2019-03-08 by echaplet

EricGIRARD35, 2019-03-10:
Bonjour echaplet, j'ai le même problème que toi quelle est la procédure pour réinitialiser les capteurs?

echaplet, 2019-03-10:
Il faut rester appuyé pendant 3 à 5 secondes sur le bouton du capteur aqara
jusqu'à ce que ça clignote plusieurs fois.

EricGIRARD35, 2019-03-12:
Je te remercie de cette réponse. C'est bien la procédure que j'utilisais mais mes contacts de porte xiaomi n'était pas reconnu. Après de nombreux essais j'ai réussi a inclure mes contact de porte qui ont fonctionné 10 mn.
Aujourd'hui plus rien ne fonctionne! mes équipement se désactive après le redémarrage du Demon.
Je regrette d'avoir changé de version, tout fonctionnait parfaitement avant...

pipiche38, 2019-03-13:
Ne serait-ce pas un problème de batterie?