fairecasoimeme/ZiGate #173: Seting Zigat in Router mode

pipiche38, 2019-03-11:
When setting the Device Type ( command 0x0023 ), we can use

> 0 = Coordinator
> 1 = Router
> 2 = Legacy Router

Could you kindly confirm that the 3 modes are possible for the Zigate ?
Could also let us know what is the difference between Legacy Router and Router ? 

So if I want to set a Zigate simply as a Router, I would only have to set Device Type to 1 or 2, correct ?

pipiche38, 2019-03-13:
En admettant qu'on puisse le configurer en Router. Il faut alors qu'il envoie un Device Annouce à un controller. Cette commande à ma connaissance n'existe pas  dans la liste des commandes Zigate.

ISO-B, 2019-03-19:
Value | Device type
------|-------------
0 | Coordinator
1 | Router
2 >= |  These values are converted to 1(Router). It should be end device, but it is disabled, because ZiGate would be very bad end device.

ISO-B, 2019-03-20:
@fairecasoimeme can you update those values to documentation.

fairecasoimeme, 2019-03-20:
fixed

closed 2019-03-20 by fairecasoimeme

dondipietro, 2019-04-10:
Hello,
How can we proceed to setup the device as a router. I tried with firmware ZiGate_v30f.bin

1) on the NXP GUI, I do erase PD
2) I click on Set Type "Router"
3) ??? when I click the start scan, I have this error :
19:52:27.641 -> 01 02 10 25 02 10 02 10 25 03 
19:52:27.727 <- 01 80 00 00 05 A2 02 00 00 25 00 03

Type: 0x8000 (Status)
  Length: 5
  Status: 0x02 (Unhandled Command)
  SQN: 0x00
  Message: 

My feeling is the firware is coordinator only, to allow router mode maybe we need the "ZigbeeNodeControlBridge_JN5168_FULL_FUNC_DEVICE"

What do you think ?
Do you succeed to connect a zigate in router mode to an other zigate in coordinator mode ?

Thank you.

pipiche38, 2019-04-10:
I suspect that is not supported by ZGUi but if you use the API that should work. 
That is something I have in mind.

dondipietro, 2019-04-12:
Thank you for your answer. Do you ever managed to use a zigate as a router ?

pipiche38, 2019-04-12:
@dondipietro , not yet, that is on my To-Do list.
But that might be the case pretty soon (I have some spare time now as I completed the implementation of an additional feature to the Domoticz Plugin) . I'll keep you posted

pipiche38, 2019-04-13:
I just did test with ZGUI and at least the commands seems to be accepted by ZGUI and Zigate
[image]

ruimarinho, 2021-03-28:
@pipiche38 is this something you could do without ZGUI?

pipiche38, 2021-03-28:
@ruimarinho I suggest that you use this firmware if you want  to use a ZiGate as a router ;-)
https://github.com/badzz/Zigate-Router

ruimarinho, 2021-09-21:
@pipiche38 I tried both binaries available but none seem targeted to JN5168. This is what I get:

```
❯ JennicModuleProgrammer -V 6 -P 115200 -e -f Router_NtagIcode_JN5169_DR1175.bin -s /dev/cu.usbserial-0001
JennicModuleProgrammer Version: 0.7(r56124)
Setting baudrate: 115200
Detected Chip: Unknown
MAC Address:   00:15:8D:00:01:ED:46:B0
Opened firmware file size 154780
Module uses Bootloader v2 Header
Incompatible firmware (Built for 0x0f03000b, Device is 0x07030008)

❯ JennicModuleProgrammer -V 6 -P 115200 -e -f Router_NtagIcode_JN5179_DR1175.bin -s /dev/cu.usbserial-0001
JennicModuleProgrammer Version: 0.7(r56124)
Setting baudrate: 115200
Detected Chip: Unknown
MAC Address:   00:15:8D:00:01:ED:46:B0
Opened firmware file size 164096
Module uses Bootloader v1 Header
Incompatible firmware (Built for 0x99887766, Device is 0x07030008)
```

I'm migrating to Z-Stack but would like to keep my ZiGate v1 around as a Router. Now that v2 is out, hopefully it won't become obsolete (cc @fairecasoimeme).

ruimarinho, 2021-09-21:
Maybe @badzz can help here too?

dmsherazi, 2022-12-14:
@any updates on this?