fairecasoimeme/ZiGate #176: How to control the ZigBee devices throw the Zigate

touzi95, 2019-03-19:
Good Morning everyone,

I am new to the ZigBee, and i found that the ZiGate solution is so interesting to my school project,
So my question is how to command the ZigBee devices throw the ZiGate is there any simple program (C/C++) of how we pair a device and command it. 
I'll be so grateful if someone helps me, or gives me his mail or something to contact him.

Thank you a lot.

ISO-B, 2019-03-19:
I don't think that there is any C or C++ libraries done, but I can be wrong. If you plan to write using C or C++ it should not be that hard. 

1. Implement serial protocol so you can talk to ZiGate. It uses baundrate 115200.
2. Implement ZiGates commanding protocol (https://zigate.fr/documentation/protocole-de-la-zigate/)
2. Send Set Channel command (0x0021)
3. Send Set Type Coordinator (0x0023)
4. Send Start Network (0x0024)
5. Send Permit join command (0x0049)
6. Pair your device

Command documentation: https://zigate.fr/wiki/commandes-zigate/
PHP Example: https://zigate.fr/documentation/exemple-api-en-php/
Doudz Python library (Quite easily readable if you need tips for your code): https://github.com/doudz/zigate

touzi95, 2019-03-19:
but the Zigate is implemented with C !!
https://github.com/fairecasoimeme/ZiGate/tree/master/Module%20Radio/Firmware/src/ZiGate/Source/ZigbeeNodeControlBridge

ISO-B, 2019-03-19:
Yes ZiGate's firmware is written in C which is then built to binaries and programmed to device. You can control ZiGate with any programming language that allows you making serial connection to ZiGate. Do you want to control ZiGate by using already implemented commands or do you want develop ZiGate firmware?

touzi95, 2019-03-19:
I want to control ZiGate by using already implemented commands, 
i am a student and as a school project, using C or C++ i have to write a little program that commands a Phillips lamp ( ON/OFF) 
so my program should read the lamp state and can send a command to change it's state.

ISO-B, 2019-03-19:
Ok, then you need to write C or C++ program that lets you to communicate with serial port so you can communicate with ZiGate. After that you implement ZiGates commanding protocol. At this point you can talk to ZiGate and it can talk back to you. Now you can send any command you want. To setup network you need to send those commands I wrote on first response. When your Phillips lamp joins network you will get it address. With that address you can send on/off command(0x0092) to toggle it on/off.

touzi95, 2019-03-19:
how to implement ZiGates commanding protocol ??

doudz, 2019-03-19:
Il y a de nombreux exemples écrit en PHP et python , il suffit de se connecter au port série pour dialoguer avec la zigate.
Toutes les informations nécessaires sont sur le github et sur le site zigate.fr

touzi95, 2019-03-19:
Can you give me a link to an example in PHP or Python ??
Thank you

doudz, 2019-03-19:
https://zigate.fr/Wiki_Categories/developpeur/

touzi95, 2019-03-19:
sorry can you give me a link for php / python example of the implementation of ZiGates commanding protocol ??

doudz, 2019-03-19:
If you can't click on a link, I can't do anything for you

touzi95, 2019-03-19:
but you just give me a link to developer page of the ZiGate web site !!!!

touzi95, 2019-03-19:
Can you give me the link to the python version, please ?? 
Thank you

ISO-B, 2019-03-19:
Link to python library and php example can be both found from my first reply: https://github.com/fairecasoimeme/ZiGate/issues/176#issuecomment-474291259

doudz, 2019-03-19:
If you want I could write your C++ program (sarcasm, sorry)

touzi95, 2019-03-19:
Looool, if you just know how to understand people's question and especially how to answer them that will be great and enough. 
Thank you anyway for your "help" (sarcasm, sorry)

doudz, 2019-03-19:
On the developper page I mention there's all you need as protocol details, php example, etc

touzi95, 2019-03-19:
I just asked for the example, and what you said ?? If you can't click on a link, I can't do anything for you. 
that's not how we answer people and we help them.

pipiche38, 2019-04-03:
@touzi95 ping me on MP (Keybase.io) I'll see how to help you

pipiche38, 2019-08-23:
can be close

closed 2019-08-23 by ISO-B