fairecasoimeme/ZiGate #179: [Feature request] Add Alarm Command

doudz, 2019-03-20:
I bought a HEIMAN siren http://www.heimantech.com/product/153.html
I paired it successfully to zigate but now I need some command to raise alarm and currently zigate doesn't provide any
The siren supports cluster 0x0009

doudz, 2019-03-20:
I think we need to support eCLD_AlarmsSignalAlarm and eCLD_AlarmsClearAlarm

ISO-B, 2019-04-05:
Can you post which clusters it supports? Does it support 0x0502? It seems more correct for this device than alert cluster.

doudz, 2019-04-05:
Extracted from my zigate.json
```json
{
            "addr": "c008",
            "discovery": "auto-discovered",
            "endpoints": [
                {
                    "clusters": [
                        {
                            "attributes": [
                                {
                                    "attribute": 0,
                                    "name": "zcl_version",
                                    "value": null
                                },
                                {
                                    "attribute": 1,
                                    "name": "application_version",
                                    "value": null
                                },
                                {
                                    "attribute": 2,
                                    "name": "stack_version",
                                    "value": null
                                },
                                {
                                    "attribute": 3,
                                    "name": "hardware_version",
                                    "value": null
                                },
                                {
                                    "attribute": 4,
                                    "data": "Heiman",
                                    "name": "manufacturer",
                                    "value": "Heiman"
                                },
                                {
                                    "attribute": 5,
                                    "data": "WarningDevice",
                                    "name": "type",
                                    "type": "str",
                                    "value": "WarningDevice"
                                }
                            ],
                            "cluster": 0
                        },
                        {
                            "attributes": [
                                {
                                    "attribute": 32,
                                    "data": 40,
                                    "name": "battery_voltage",
                                    "type": "float",
                                    "value": 4.0
                                },
                                {
                                    "attribute": 33,
                                    "data": 192,
                                    "name": "battery_percent",
                                    "type": "int",
                                    "value": 192
                                },
                                {
                                    "attribute": 65533
                                }
                            ],
                            "cluster": 1
                        },
                        {
                            "attributes": [
[... 116 lines cut ...]
                        3,
                        4,
                        9,
                        1280,
                        1282
                    ],
                    "out_clusters": [
                        3,
                        25
                    ],
                    "profile": 260
                }
            ],
            "generictype": "",
            "info": {
                "addr": "c008",
                "id": 41,
                "ieee": "000d6f00110888eb",
                "last_seen": "2019-04-05 14:15:28",
                "lqi": 132,
                "mac_capability": "10001110",
                "power_type": 1
            }
        },
```

ISO-B, 2019-04-16:
I have sent test firmware for doudz for testing. I will report back after testing.

ISO-B, 2019-07-05:
I received my siren. After testing it seems that it supports Squawk from IAS Warning Device Cluster, but I haven't yet managed to successfully use start warning command. For squawk device seems to only support one squawk level.

ISO-B, 2019-07-05:
Managed to get Start warning command work. I will make PR at Monday or Tuesday.

pipiche38, 2019-07-05:
But this is already addressed with Firware 3.1a, i have the same and the warning works perfectly.
@akila made it

pipiche38, 2019-07-05:
```
Zigate -> Obj  0x0111  Write Attribute request IAS_WD (from v3.1a)
                       <address mode: uint8_t>
   <target short address: uint16_t>
   <source endpoint: uint8_t>
   <destination endpoint: uint8_t>
   <direction: uint8_t>
   <manufacturer specific: uint8_t>
   <manufacturer id: uint16_t>
   <Warning Mode: uint8_t>
   <Warning Duration: uint16_t>
   <Strobe duty cycle : uint8_t>
   <Strobe level : uint8_t>

```

pipiche38, 2019-07-05:
Now, implementing the Squawk command woulld be great too

ISO-B, 2019-07-05:
Ah, didn't notice that, because I haven't followed progress very closely in summer. I will add squawk.

pipiche38, 2019-07-05:
Ah ah the finish summer time ;-)

pipiche38, 2019-07-14:
This can be closed as it is fully addressed in 3.1a

closed 2019-07-15 by ISO-B