fairecasoimeme/ZiGate #181: Error Zigate a repetition

sifus, 2019-03-30:
Hello,

mon log zigate est remplie de la meme erreur.

```
[2019-03-29 23:32:46][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:43d7, sequence:179, lqi:0
[2019-03-30 00:03:18][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:2229, sequence:180, lqi:0
[2019-03-30 00:03:49][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:1dd9, sequence:181, lqi:0
[2019-03-30 00:04:29][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:a79a, sequence:182, lqi:0
[2019-03-30 00:33:08][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:43d7, sequence:183, lqi:0
[2019-03-30 01:03:39][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:2229, sequence:184, lqi:0
[2019-03-30 01:04:05][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:1dd9, sequence:185, lqi:0
[2019-03-30 01:04:41][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:a79a, sequence:186, lqi:0
[2019-03-30 01:33:22][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:43d7, sequence:187, lqi:0
[2019-03-30 02:04:01][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:2229, sequence:188, lqi:0
[2019-03-30 02:04:22][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:1dd9, sequence:189, lqi:0
[2019-03-30 02:04:54][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:a79a, sequence:190, lqi:0
[2019-03-30 02:33:42][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:43d7, sequence:191, lqi:0
[2019-03-30 03:04:20][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:2229, sequence:192, lqi:0
[2019-03-30 03:04:37][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:1dd9, sequence:193, lqi:0
[2019-03-30 03:05:03][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:a79a, sequence:194, lqi:0
[2019-03-30 03:34:04][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:43d7, sequence:195, lqi:0
[2019-03-30 04:04:44][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:2229, sequence:196, lqi:0
[2019-03-30 04:04:52][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:1dd9, sequence:197, lqi:0
[2019-03-30 04:05:15][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:a79a, sequence:198, lqi:0
[2019-03-30 04:34:25][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:43d7, sequence:199, lqi:0
[2019-03-30 04:56:14][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:2229, sequence:200, lqi:0
[2019-03-30 05:05:09][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:1dd9, sequence:201, lqi:0
[2019-03-30 05:05:30][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:a79a, sequence:202, lqi:0
[2019-03-30 05:34:47][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:43d7, sequence:203, lqi:0
[2019-03-30 05:56:36][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:2229, sequence:204, lqi:0
[2019-03-30 06:05:27][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:1dd9, sequence:205, lqi:0
[2019-03-30 06:05:45][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:a79a, sequence:206, lqi:0
[2019-03-30 06:35:09][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:43d7, sequence:207, lqi:0
[2019-03-30 06:56:57][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:2229, sequence:208, lqi:0
[2019-03-30 07:05:43][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:1dd9, sequence:209, lqi:0
[2019-03-30 07:06:00][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:a79a, sequence:210, lqi:0
[2019-03-30 07:35:33][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:43d7, sequence:211, lqi:0
[2019-03-30 07:57:19][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:2229, sequence:212, lqi:0
[2019-03-30 08:05:59][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:1dd9, sequence:213, lqi:0
[2019-03-30 08:06:12][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:a79a, sequence:214, lqi:0
[2019-03-30 08:35:48][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:43d7, sequence:215, lqi:0
[2019-03-30 08:57:38][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:2229, sequence:216, lqi:0
[2019-03-30 09:06:14][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:1dd9, sequence:217, lqi:0
[2019-03-30 09:06:27][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:a79a, sequence:218, lqi:0
[2019-03-30 09:36:11][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:43d7, sequence:219, l [... cut ...]
[2019-03-30 17:08:18][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:1dd9, sequence:250, lqi:0
[2019-03-30 17:17:08][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:a79a, sequence:251, lqi:0
[2019-03-30 17:38:31][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:43d7, sequence:252, lqi:0
```

ISO-B, 2019-04-01:
Do you have more than 8 devices paired to your ZiGate? All devices work well? What kind of devices you have?

Status code 240 is 0xF0. Chapter 10.2.4 Mac codes in JN-UG-3113 v1.5 states:

    0xF0 Pending transaction has expired and data discarded

This means that node you are trying to send commands didn't receive command within time limit

sifus, 2019-04-01:
Hi,
i have 11 device and 16 soon

[url=http://zupimages.net/viewer.php?id=19/14/kxfc.png][img]https://zupimages.net/up/19/14/kxfc.png[/img][/url]

zigdir, 2019-04-01:
But all of your sensors seems to be battery based. I don't think that they can accept any inbound communication outside of the Pairing process.

pipiche38, 2019-08-22:
What I have observe and which is also very strange, is that the Zigate is sending APS Failure not in relation to any command send to Zigate.
In other terms we are receiving APS Failure most-likely for internal matters ! And that shouldn't be possible as what the above layer should do on pure internal mechanism on lower level

pipiche38, 2020-09-30:
This issue has been closed due to no activity since while!

closed 2020-09-30 by fairecasoimeme