fairecasoimeme/ZiGate #182: Flash PiZigate - Error setting baudrate

GPierrot, 2019-03-31:
Bonjour,

Après réception de ma PiZigate, j’ai voulu la mettre a jour en suivant les instructions du Wiki.

A la première utilisation de JennicModuleProgrammer, l’operation s’est bien lancée, mais a planté sur une erreur ´Failed to write at address‘.
Depuis, a chaque nouvelle tentative, je recois le message suivant :
JennicModuleProgrammer version: 0.6 (r56124)
Setting baudrate: 115200
Error setting baudrate
🥺

Merci par avance pour vos idées ou solutions 😬

Martial83, 2019-04-04:
J'ai exactement le même problème, mais sans être passé par la case "Failed to write at address" mais j'ai pu résoudre mon problème initial sans reflasher, mais en faisant un erase persistant data.

ISO-B, 2019-04-04:
I have Raspberry Pi 2B with Raspbian. I had to remove console=serial0,115200 from /boot/cmdline.txt before flashing started working. I used https://github.com/tjikkun/zigate-flasher with following command: `sudo python3.5 zigate-flasher.py -p /dev/ttyAMA0 -w /home/pi/firmwares/firmware.bin`

Martial83, 2019-04-04:
Thank ISO-B
I have already suppress console=serial0,115200 from /boot/cmdline.txt.
The problem seems to be near JennicModuleProgrammer.
I will try zigate-flasher soon

GPierrot, 2019-04-04:
> I have Raspberry Pi 2B with Raspbian. I had to remove console=serial0,115200 from /boot/cmdline.txt before flashing started working. I used https://github.com/tjikkun/zigate-flasher with following command: `sudo python3.5 zigate-flasher.py -p /dev/ttyAMA0 -w /home/pi/firmwares/firmware.bin`

I tried this, but I have again an error message related to baudrate :
[image]

GPierrot, 2019-04-04:
Just to be clear on the process 
- config.txt is changed
- cmdline.txt is changed
- GPIO control is installed

result of a new test just here after
[image]

Martial83, 2019-04-04:
Sorry GPierrot,
Are you using zigate-flasher ou JennicModuleProgrammer ?

GPierrot, 2019-04-05:
I tried both :

Initialy, I used Jennic and I received the baud rate error message.
Following Iso-b feedback, I tried zigate-flasher without success.

ISO-B, 2019-04-05:
Have you switched between production and flashing pins between tries?
I have following sh scripts that I use to toggle between modes, before and after programming. Also sometimes booting before flashing can fix things. If nothing else works and you have usb to ttl adapter available you could try programming chip using it.

pizigate_flash.sh
```bash
#!/bin/bash

gpio mode 0 out
gpio mode 2 out
gpio write 2 0
gpio write 0 0
gpio write 0 1
```

pizigate_production.sh
```bash
#!/bin/bash

gpio mode 0 out
gpio mode 2 out
gpio write 2 1
gpio write 0 0
gpio write 0 1
```

pipiche38, 2019-04-05:
Here is the script which is part of the Domoticz Zigate Plugin and which can be used for seting the PiZigate in Run or Flash mode

https://github.com/sasu-drooz/Domoticz-Zigate/blob/stable/Tools/pi-zigate.sh

fairecasoimeme, 2019-04-06:
Hi,
I have just developed a tool to test if the PiZiGate works.
You can find this here :
[https://github.com/fairecasoimeme/ZiGate/tree/master/Tools/PiZiGate](https://github.com/fairecasoimeme/ZiGate/tree/master/Tools/PiZiGate)

Here an example when all it's OK.
[image]

nunus59147, 2019-04-08:
with a friend we both have the same issue

GPierrot, 2019-04-09:
Update :

I used the testing tool, with the following result : 
[image]

Following the test, the flash was successful using JennicModuleProgrammer :  
[image]

I used again the testing tool : 
[image]

Good progress, thanks for your help.
My remaining concern is that the version looks not correct : 0180

fairecasoimeme, 2019-04-09:
Hi,
It's possible that datas are not good because I do very too simple interpretation. The aim was to show bidirectional communication.
I'll evolve the tools rapidly :)

pipiche38, 2020-09-30:
This issue has been closed due to no activity since while!

closed 2020-09-30 by fairecasoimeme

xlyric, 2021-07-17:
voulu faire une mise à jour ce jour et même problème 

`/boot/cmdline.txt
dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=00690444-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
`
`root@raspberrypi:~/1/PiZiGate-tools/test# ./PiZiGate_test /dev/ttyAMA0
RPI Model searching ...
Warning !!! you must modify /boot/config.txt
Verifying /boot/cmdline.txt ...
+ /boot/cmdline.txt seems to be OK
Verifying /boot/config.txt ...
+ /boot/config.txt seems to be OK
Warning !!! if it's not the case, you have to execute the following commands :
sudo systemctl disable hciuart
sudo usermod -aG gpio pi
then reboot the PI
Searching /dev/ttyAMA0...
+ Port : /dev/ttyAMA0 is not used
Verif GPIOs ...
+ GPIO 0 (RESET) --> OK
+ GPIO 2 (FLASH) --> OK
Config GPIOs ...
+ GPIO 0 (RESET) --> OK
+ GPIO 2 (FLASH) --> OK
Opening : /dev/ttyAMA0 ...
+ /dev/ttyAMA0 opened --> OK
+ Packet 01 02 10 10 02 10 02 10 10 03 sent --> OK
+ Packet received --> OK
size : 82
 01 80 24 00 0d 7c 01 00 00 00 15 8d 00 01 ed e9 42 0b 00 03
01 80 06 00 02 85 01 00 03
01 80 00 00 05 95 00 00 00 10 00 03
01 80 10 00 05 9a 00 03 03 0f 00 03
`

et en mode update : 
`./JennicModuleProgrammer -V 255 -P115200  -f "ZiGate_v3.1e.bin" -s /dev/ttyAMA0
JennicModuleProgrammer Version: 0.7(r56124)
Setting baudrate: 115200
Error setting baudrate
`

pas de solution trouvée sur le net avant de poster

xlyric, 2021-07-17:
étrangement c'est passé ... à moitié
j'ai remis en mode prod, 
redémarré domoticz
arreté domoticz
remis en mode update et relancé le script 

`oot@raspberrypi:~/1/JennicModuleProgrammer/Build# ./JennicModuleProgrammer -V 6  -P115200  -f "ZiGate_v3.1e.bin" -s /dev/ttyAMA0
JennicModuleProgrammer Version: 0.7(r56124)
Setting baudrate: 115200
Detected Chip: Unknown
MAC Address:   00:15:8D:00:01:ED:E9:42
Opened firmware file size 208404
Module uses Bootloader v2 Header
Erasing:   100%
Writing Program to Flash
Writing:   100%
Verifying Program in Flash
Verifying:  89%
Error reading at address 0x0002db80
`
et impossible de relancer le script ensuite 
```
JennicModuleProgrammer Version: 0.7(r56124)
Setting baudrate: 115200
Error setting baudrate

```

par contre en redémarrant domoticz il a quand même pris la nouvelle version