fairecasoimeme/ZiGate #183: Over-The-Air command

zigdir, 2019-04-01:
I'm new to this , and maybe I didn't look the right place, where could I find the information to get Over-The-Air command ?

Did look at https://zigate.fr/wiki/commandes-zigate/ , but did not see any references to OTA

Does that mean Zigate do not support OTA ?

ISO-B, 2019-04-01:
Hi,
ZiGate supports OTA, but it seems that those commands are missing from documentation.
Commands are:

Command | Code | Description
-----------|------|------------
LOAD_NEW_IMAGE | 0x0500 | Load headers to ZiGate. Use this command first.
BLOCK_REQUEST | 0x8501 | ZiGate will receive this command when device asks OTA firmware
BLOCK_SEND | 0x0502 | This is used to transfer firmware to device when it sends request 0x8501. 
UPGRADE_END_REQUEST | 0x8503 | Device will send this when it has received last part of firmware
UPGRADE_END_RESPONSE | 0x0504 | 
IMAGE_NOTIFY | 0x0505 | Notify desired device that ota is available. After loading headers use this.
SEND_WAIT_FOR_DATA_PARAMS | 0x0506 | Can be used to delay/pause OTA update

If you just need to update something using OTA you could use [TestGUI](https://github.com/fairecasoimeme/ZiGate/tree/master/Module%20Radio/Firmware/src/ZiGate/Tools/TestGUI/TestGUI)(Windows only) or [doudz python library](https://github.com/doudz/zigate#ota-update) which has ota implemented. Other libraries might have also OTA implemented, but I have only been using doudz. I don't have time right now to write full description of every command and required payload, but will do it tomorrow.

zigdir, 2019-04-02:
Could you let me know where I can find the API description of those command ? For instance the number of fileds and field description.
From the ZGUI , I have found the OTA Cluster and it works. However it looks from the Raw Data (in ZGUI) that they are some expected fields after Header Str

zigdir, 2019-04-02:
Ok. found I think what I was looking for in the ZiGate source code app_Znccmds.c

pipiche38, 2019-04-05:
I would also recommend you the issue #86 which give a good documentation on the upgrade OTA
https://github.com/fairecasoimeme/ZiGate/issues/86#issuecomment-447884154

pipiche38, 2019-04-08:
Just would like to comment, that we have here a LOAD_NEW_IMAGE, but I think it would be beneficial to have LIST_IMAGES in order to get the list of images headers loaded into the Zigate , otherwise I don't see any control

ISO-B, 2019-04-08:
There is currently no need for such command since ZiGate can only have one header loaded at any given time. So if you have previously loaded header and you load another header old header is overwritten.

pipiche38, 2019-04-08:
Any possibility to have the Zigate capable to store more than only 1 Image. This could really help in terms of Load, and then use the JITTER flag accordingly