fairecasoimeme/ZiGate #184: Support modules Legrand à base de EM2260 (non Netatmo)

Martial83, 2019-04-05:
Bonjour et un grand merci pour le travail réalisé.
Je possède chez moi des modules Legrand d'avant la fusion avec Netatmo: un interrupteur radio double et 2 relais.
C'est du zigbee (à base de EM260 ) et je souhaite pouvoir les gérer avec ma PiZigate (et Domoticz).
Ai-je une chance d'y arriver ?
Cordialement

fairecasoimeme, 2019-04-06:
Peux-tu donner les références du matériel.
Si c'est du MyHome, j'ai déjà essayé mais malheureusement c'est du Zigbee propriétaire.

Martial83, 2019-04-06:
J'ai du mal à suivre et à savoir sir c'est du MyHome ou pas:
Module interrupteur 088206 exemple: https://www.confort-electrique.fr/notices/notice-legrand-88206.pdf

Mais c'est peut-être bien du MyHome ... snif...

Martial83, 2019-04-07:
Bon, j'ai dis une grosse bêtise ... ce ne sont pas les modules indiqués dans mon poste précédent mais:
Interrupteur radio: 088306
Commutateur double: 067224
Ce qui ne change pas grand chose: ça semble être du MyHome Play!

pipiche38, 2019-07-17:
Je pense qu'on peut clore cet issue

closed 2019-07-17 by fairecasoimeme

Martial83, 2019-07-17:
Oui, on peut clore, mais c'est dommage (pour moi ?)