fairecasoimeme/ZiGate #185: TestGUI et PiZigate

Martial83, 2019-04-05:
Bonjour,
Comment utiliser testGUI avec une PiZigate, qui, par définition n'est pas sous Windows ?
Je ne trouve pas d'alternative à testGUI qui fonctionne sous Debian (ou autre Linux)

Une idée ? Une piste ?

pipiche38, 2019-04-05:
@Martial83 c'est une très bonne question. Avec le Plugin Zigate (Domoticz) tu trouveras en plus du script pi-zigate.sh pour piloter le Gpio, un script python SerialLink.py

Il est en fait disponible dans les Sources de la Zigate , mais avec pas mal de Bug. 
https://github.com/fairecasoimeme/ZiGate/tree/master/Module%20Radio/Firmware/src/ZiGate/Tools

fairecasoimeme, 2019-04-06:
Hi,
I have just developed a tool to test if the PiZiGate works.
You can find this here :
[https://github.com/fairecasoimeme/ZiGate/tree/master/Tools/PiZiGate](https://github.com/fairecasoimeme/ZiGate/tree/master/Tools/PiZiGate)

Martial83, 2019-04-06:
Bonjour,
Pour être précis, mon PiZigate fonctionne maintenant parfaitement avec mes différents appareils Xiaomi (boitier interrupteur rond, détecteur temperature/humidité, détecteurs de présence Aqara. Donc le test n'est pas utile dans mon cas, mais peut certainement en aider beaucoup.
Je cherche à savoir si je vais pouvoir intégrer mes équipement Legrand Celiane Zigbee (mais pas Netatmo)
Pour éviter les doublons, on continue sur l'issue#184?

pipiche38, 2020-09-30:
This issue has been closed due to no activity since while!

closed 2020-09-30 by fairecasoimeme