fairecasoimeme/ZiGate #187: Plugin ZIGATE lancement des dépendances Impossible

advence87, 2019-04-12:
Bonjour à tous,
Ma clef fraichement reçu avant-hier fait des siennes à moins que le probleme vienne du plugin. 
Impossible d'installer les dépendances pas de version lib et pas de version de version firmware au-dessus de la clef et le probléme persiste avec la version beta du plugin zigate. 
Si une âme charitable à un debut de solution cela m 'interresse et bien entendu j'ai ouvert un ticket sur jeedom avec une jolie réponse de m'adresser au developpeur du plugin. :mrgreen: 

Configuration jeedom
Jeedom smart à jour 3.2.16 
pip2 -V
pip 18.0 from /usr/local/lib/python2.7/dist-packages/pip (python 2.7)
pip3 -V
sh: 1: pip3: not found
pip -V
pip 18.0 from /usr/local/lib/python2.7/dist-packages/pip (python 2.7)
plugin Zigate 1.2.3
Comment mettre à jour la Lib Python 3 sur la jeedom smart?
Merci de votre aide

[zigate_update.txt](https://github.com/fairecasoimeme/ZiGate/files/3074108/zigate_update.txt)
[zigate2.log](https://github.com/fairecasoimeme/ZiGate/files/3074109/zigate2.log)

fairecasoimeme, 2019-04-13:
Désolé mais je pense qu'il faudrait poster le problème ici :
[https://github.com/Jeedom-Zigate/jeedom-plugin-zigate/issues](https://github.com/Jeedom-Zigate/jeedom-plugin-zigate/issues)

pipiche38, 2019-07-25:
should be closed as not the right place

doudz, 2019-07-25:
Yes could be closed (fixed for months)

closed 2019-07-25 by ISO-B