fairecasoimeme/ZiGate #188: PiZiGate test erreur

oasisck, 2019-04-13:
Bonjour, PiZiGate fraîchement reçu et inséré dans un Raspberry 3 avec un jeedom fresh install.
Impossible de faire fonctionner la PiZiGate. J'ai suivi les tutos :
- https://zigate.fr/documentation/compatibilite-raspberry-pi-3-et-zero-w/
- https://zigate.fr/documentation/parametrer-la-pizigate/
Actuellement la lumière est bleue forte (mode production). Le plugin Zigate et Abeille n'arrive pas à faire fonctionner la PiZiGate.
Voilà le résultat de test : 
```
root@jeedom:~/Test/Bin# ./PiZiGate_test /dev/ttyS0
Searching /dev/ttyS0...
+ Port : /dev/ttyS0 is not used
Verif GPIOs ...
+ GPIO 0 (RESET) --> OK
+ GPIO 2 (FLASH) --> OK
Config GPIOs ...
+ GPIO 0 (RESET) --> OK
+ GPIO 2 (FLASH) --> OK
Opening : /dev/ttyS0 ...
+ /dev/ttyS0 opened --> OK
- Error from tcgetattr: Input/output error
- Error from write: -1, 5
- No packet received - size sent : -1 - error : 5
```
Branchement de la PiZiGate vérifié plusieurs fois.
Je ne sais plus où chercher.
Merci de votre aide

fairecasoimeme, 2019-04-13:
pouvez-vous me copier les deux fichiers :

/boot/config.txt
/boot/cmdline.txt

oasisck, 2019-04-15:
[Résolu](https://zigate.fr/forum/topic/pizigate-aucune-donnee/?part=2)
ttyAMA0 était tenu par la console serial de Raspbian (peut être activé via le script d'installation de Jeedom)
Pour désactiver :
systemctl stop serial-getty@ttyAMA0.service
systemctl disable serial-getty@ttyAMA0.service
```
root@jeedom:~/Test/Bin# ./PiZiGate_test /dev/ttyAMA0
Searching /dev/ttyAMA0...
+ Port : /dev/ttyAMA0 is not used
Verif GPIOs ...
+ GPIO 0 (RESET) --> OK
+ GPIO 2 (FLASH) --> OK
Config GPIOs ...
+ GPIO 0 (RESET) --> OK
+ GPIO 2 (FLASH) --> OK
Opening : /dev/ttyAMA0 ...
+ /dev/ttyAMA0 opened --> OK
+ Packet received --> OK
Release : 0003
Version : 030f
```
Merci pour le support sur le forum 👍

closed 2019-04-15 by oasisck